mne-tools/mne-python #7715: BUG: VectorSourceEstimate.plot() in v20 does not show the surfaces

yh-luo, 2020-05-03:
#### Describe the bug
`VectorSourceEstimate.plot()` is not working properly in v20.

#### Steps to reproduce
I used the `sample` dataset to compute some `VectorSourceEstimate` files to play around. Trouble shootings in [Advanced setup of MNE-Python](https://mne.tools/stable/install/advanced.html#advanced-setup-of-mne-python) did not fix the problem.
```python
import os.path as op
import mne
from scripts.config import baseline, meg_dir, subjects_dir

os.environ["SUBJECTS_DIR"] = subjects_dir
stc_aud = mne.read_source_estimate(
    op.join(meg_dir, 'aud_left_minus_right-grand_average-ave-stc'))
stc_aud.plot(hemi='both', initial_time=0.1)
```

#### Expected results
In MNE v19, it works without problems
[image]

#### Actual results
In MNE v20, the surfaces disappear after plotting.
[image]

The problem persists in `master`.

#### Additional information
If required, I can upload the source estimate file.

`mne.sys_info`
```
Platform:      Linux-5.6.8-200.fc31.x86_64-x86_64-with-fedora-31-Thirty_One
Python:        3.7.7 (default, Mar 13 2020, 10:23:39)  [GCC 9.2.1 20190827 (Red Hat 9.2.1-1)]
Executable:    /home/yuhan/env/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        15.5 GB

mne:           0.20.3
numpy:         1.17.3 {blas=openblas, lapack=openblas}
scipy:         1.3.2
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.21.3
numba:         0.46.0
nibabel:       2.5.1
cupy:          Not found
pandas:        0.25.3
dipy:          1.0.0
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.13.2}
pyvista:       0.24.1
vtk:           8.1.2
```

larsoner, 2020-05-03:
This could be garbage collection, if you do `brain = stc.plot(....)` to keep a reference to the object, does it help?

yh-luo, 2020-05-03:
@larsoner No, ```brain = stc_aud.plot(hemi='both', initial_time=0.1)``` shows the same behavior. Also, the bug exists in both Python interpreter and IPython.

yh-luo, 2020-05-03:
I used [Visualize source time courses](https://mne.tools/stable/auto_tutorials/source-modeling/plot_visualize_stc.html#sphx-glr-auto-tutorials-source-modeling-plot-visualize-stc-py) to further test `SourceEstimate.plot()`, `VolSourceEstimate.plot()` and `VectorSourceEstimate.plot()`, only `VectorSourceEstimate` has this problem.

yh-luo, 2020-05-03:
I think depth peeling could be the issue. The surfaces are shown correctly (but ugly) when `brain_alpha=1`. I tried to comment out the below part to avoid depth peeling and the problem is fixed too.

https://github.com/mne-tools/mne-python/blob/5f4c9046fb07cf75a316d56d809e4b7266753ceb/mne/viz/_3d.py#L2503-L2507

yh-luo, 2020-05-04:
I'm closing this because it's actually a problem of Fedora with NVIDIA Optimus laptop. Others may encounter the same Mayavi rendering [problem](https://stackoverflow.com/questions/52621431/mayavi-not-rendering-solid-surface-correctly) so the solution is below:
1. Install the proprietary nvidia driver
2. [Configure graphics applications to render using the NVIDIA GPU](https://rpmfusion.org/Howto/Optimus). Set the environment variable `__NV_PRIME_RENDER_OFFLOAD=1` and `__GLX_VENDOR_LIBRARY_NAME=nvidia` before running graphical applications. Or set them in `.bashrc`
```bash
export __NV_PRIME_RENDER_OFFLOAD=1
export __GLX_VENDOR_LIBRARY_NAME=nvidia
```
I'm not sure if other Linux users have this problem with dual GPU laptops though.

closed 2020-05-04 by yh-luo

GuillaumeFavelier, 2020-05-04:
Thanks for sharing!

> I'm not sure if other Linux users have this problem with dual GPU laptops though.

I used a [slightly more complicated method](https://forum.manjaro.org/t/howto-set-up-prime-with-nvidia-proprietary-driver/40225) on my Manjaro to get rid of those graphical artifacts.