mne-tools/mne-python #7716: BUG: Not able to read .mat file via mne.io.read_raw_fieldtrip()

Ferdodonna, 2020-05-03:
#### Describe the bug
Two errors are raised while trying to read a .mat file. File can be opened and plotted in Matlab: 

[image]

#### Steps to reproduce
Dataset used: [P300S01.mat](https://www.dropbox.com/s/lzr3p5yuhodvbdq/P300S01.mat?dl=0) [size 40MB]

```
import mne
fname = 'C:\\Users\\...\\P300S01.mat' # file location
raw = mne.io.read_raw_fieldtrip(fname,info=None)
```

#### Expected results
I expected a Raw object to be created, using the raw variable as reference. 

#### Actual results
Actual output:

```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
~\Anaconda3\lib\site-packages\mne\io\fieldtrip\utils.py in _set_sfreq(ft_struct)
    175     try:
--> 176         sfreq = ft_struct['fsample']
    177     except KeyError:

KeyError: 'fsample'

During handling of the above exception, another exception occurred:

KeyError                                  Traceback (most recent call last)
~\Anaconda3\lib\site-packages\mne\io\fieldtrip\utils.py in _set_sfreq(ft_struct)
    178         try:
--> 179             t1 = ft_struct['time'][0]
    180             t2 = ft_struct['time'][1]

KeyError: 'time'

During handling of the above exception, another exception occurred:

ValueError                                Traceback (most recent call last)
<ipython-input-18-a190d341032d> in <module>
      3 import mne
      4 fname = 'C:\\Users\\suuuj\\Desktop\\lecture\\p300 sampling dataset\\P300S01.mat'
----> 5 raw = mne.io.read_raw_fieldtrip(fname,info=None)

~\Anaconda3\lib\site-packages\mne\io\fieldtrip\fieldtrip.py in read_raw_fieldtrip(fname, info, data_name)
     55     _validate_ft_struct(ft_struct)
     56 
---> 57     info = _create_info(ft_struct, info)  # create info structure
     58     data = np.array(ft_struct['trial'])  # create the main data array
     59 

~\Anaconda3\lib\site-packages\mne\io\fieldtrip\utils.py in _create_info(ft_struct, raw_info)
     42         warn(NOINFO_WARNING)
     43 
---> 44     sfreq = _set_sfreq(ft_struct)
     45     ch_names = ft_struct['label']
     46     if raw_info:

~\Anaconda3\lib\site-packages\mne\io\fieldtrip\utils.py in _set_sfreq(ft_struct)
    182             sfreq = 1 / difference
    183         except KeyError:
--> 184             raise ValueError('No Source for sfreq found')
    185     return sfreq
    186 

ValueError: No Source for sfreq found
```

#### Additional information

Platform:      Windows-10-10.0.18362-SP0
Python:        3.7.3 (default, Mar 27 2019, 17:13:21) [MSC v.1915 64 bit (AMD64)]
Executable:    C:\Users\suuuj\Anaconda3\python.exe
CPU:           Intel64 Family 6 Model 142 Stepping 10, GenuineIntel: 8 cores
Memory:        7.9 GB

mne:           0.20.0
numpy:         1.18.3 {blas=C:\\projects\\numpy-wheels\\numpy\\build\\openblas_info, lapack=C:\\projects\\numpy-wheels\\numpy\\build\\openblas_lapack_info}
scipy:         1.2.1
matplotlib:    3.2.0 {backend=Qt5Agg}

sklearn:       0.20.3
numba:         0.43.1
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.3
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.14.1}
pyvista:       Not found
vtk:           8.1.2

agramfort, 2020-05-03:
@thht it's a fieldtrip dataset. Any idea?

thht, 2020-05-04:
hi @Ferdodonna,
first of all, thank you for the good and detailed bug report.

from the matlab screenshot of the structure you provide it seems that this is not a FieldTrip data structure you are trying to load. the `read_*_fieldtrip` functions can only handle data structures created by FieldTrip. it might be the case that FieldTrip has changed their structures very recently, but so far i am not aware of that.

what matlab toolbox does the data structure originate from?

Ferdodonna, 2020-05-05:
Oh that is my bad. I told @agramfort it might be fieldtrip, even though I've never used Matlab. This is not my data, I just wanted to play around with P300s in MNE. I'm currently asking the author what they used for preprocessing the data.  Should I delete this issue?

thht, 2020-05-05:
no problem. however, it is quite easy to import this data into mne.

you basically load the mat file into python using pymatreader like this:

```
from mne.externals.pymatreader import read_mat
mat_data = read_mat(fname)
```

and then you use `mne.create_info` to create the `Info` object and with that create a mne datastructure from an array using one of these classes: https://mne.tools/stable/python_reference.html#id4

closed 2020-05-05 by agramfort

AiCxlai, 2022-06-11:
Hello, is this problem solved now? I also want to import the. Mat file into python program，thanks @Ferdodonna