mne-tools/mne-python #7718: import code for optical imaging data from BOXY, fNIRS and fast signal

kylemath, 2020-05-03:
#### Describe the problem
Want to create io functionality for BOXY files created for the Imagent and Imagent 2 from ISS, which collect both overall and modulated light levels, but also phase delay measures used to measure fast scattering changes in neural tissue.

#### Describe your solution

Working on a set of io functions similar to the NIRx.py options to load files in as Raw objects here: 
Loads digitization files, lists of channels used, data files, parses files and creates Raw object with dig and chan locs info, coregisters fiducials with common space, markers from marker channel, and data channel names based on location, data type, and wavelength:

https://github.com/mne-tools/mne-python/pull/7717

#### Describe possible alternatives
Import preprocessed averaged files directly from https://code.google.com/archive/p/p-pod/, coregister with existing tools in that same repo (coreg), and load in talaraich locatoins directly

#### Additional context
Some important discussions to have regarding

(These are checkboxes in the pull request)
-  data types (3 data types, overall light (DC), modulated light (AC), and phase delay (Ph), from each channel)
-  Wavelength
-  Modulation frequency - new imagent 2 machine has 4 modulation frequencies of light, which are used to measure AC and Ph in the measured signal, and may be relevant for light modeling and analysis
-  Channel names - are based on a letter/number grid on our headgear, but need to be flexible
-  digitization and coregistration - common head, individual heads, fiducial alignment, other head points, sensor locations/midpoints
-  markers from data files
-  importing multiple files per participant (old systems use on file for each detector), all files use one file for each montage (arrangement of sources and detectors), usually collected in separate sessions, for both these cases only a single digitization file is needed, but different files of source and detector locations used.
-  renaming files - BOXY outputs time stamped files, and usually matlab script is used to rename files by subject number, experiment, etc., build into this?
-  preprocessing functions - heart beat removal - regression based heart beat removal for fast signal needed as pulse wave spectra overlaps with fast signals (Corballis and Gratton) (ppod software - https://code.google.com/archive/p/p-pod/)
-  preprocessing functions - phase wrapping correction - phase signal is modified to a continuous signal (ppod software - https://code.google.com/archive/p/p-pod/)

References:
boxy optical imaging workflow:
http://sites.psych.ualberta.ca/kylemathewson/optical-imaging/

rob-luke, 2020-05-03:
I am going to comment directly on the pull request as the dot points above are the same as the checklist there. Let me know if its preferable for me to comment here instead.

larsoner, 2021-02-19:
We have this from #7717, closing

closed 2021-02-19 by larsoner