mne-tools/mne-python #7719: Add option to add vertical jitter to events plot

hoechenberger, 2020-05-04:
#### Describe the problem
When extracting the events from the first run of the [`ds000246` OpenNeuro datasat](https://github.com/OpenNeuroDatasets/ds000246) and plotting them via `mne.viz.plot_events()`, I was convinced that the number plotted `button` events didn't match the number of events in the legend. Closer inspection revealed that there was simply some hard-to-spot overplotting.

[image]

#### Describe your solution
I therefore propose to add a `jitter` parameter to `plot_events()` which would add a vertical jitter to the event data points, reducing overplotting.

agramfort, 2020-05-04:
no objection

>

larsoner, 2020-05-04:
My first thought was that we could just do `alpha=0.5` or so, as it will make it clear when there are overlaps. But I agree this still isn't as useful as making use of the vertical space.

With jitter -- which I assume means random noise added to the vertical values -- I'd worry about people interpreting it as meaningful somehow. Rather than jitter, how about we actually make it meaningful/helpful and to each `these_events` corresponding to some integer event value do:
```
y = this_event_number + (np.arange(len(these_event_times)) % 10) * 0.5
```
This will make all events of type 1 for example straircase upward for 10 values, then drop back down. This is:

1. non-random
2. facilitates interpretation, because you can more easily see for example that something is the 33rd event, because it'll be the third one on the fourth staircase
3. maintains a gap of 0.5 between event numbers to help keep it clear that they are still integers

WDYT?

larsoner, 2020-05-04:
(BTW I think this should be controllable with a `staircase=True` default kwarg, so that people can do `False` to get the old behavior back if they want it)

drammock, 2020-05-04:
Another possible solution would be to plot these as thin, short vertical lines like a rugplot. No strong feeling whether this is better or worse than staircase (though I like staircase better than jiitter).

agramfort, 2020-05-04:
tiny vertical bars would work too. Whatever is simple enough to do the job

>