mne-tools/mne-python #7723: Morphing label-based source estimates

MerlinDumeur, 2020-05-04:
Dear MNE team,

I would like to morph my label source estimates to an average brain, but it does not seem possible with the current mne-python API.

I have reconstructed surface sources using the `mne.minimum_norm.apply_inverse_raw` function by passing the `label` parameter.
When I then try to use `mne.SourceMorph.apply` on the source estimates, I get the following error:

> ValueError: vertices do not match between morph (10242) and stc (149) for the left hemisphere:

Evidently it is trying to match the vertices from the label to the whole hemisphere.
I tried different approaches, but I wasn't able to make it work.

I am however able to use the `mne.morph_labels` function to morph the labels prior to performing the source reconstruction, but it requires me to have two versions of the source estimates, which doubles the footprint on the disk.

Is it possible to perform the morphing after the label-based source reconstruction ?

larsoner, 2020-05-04:
Perhaps https://mne.tools/stable/generated/mne.SourceEstimate.html#mne.SourceEstimate.expand, then morph?

MerlinDumeur, 2020-05-05:
I tried using `mne.SourceEstimate.expand`, but I ran out of memory.

I should've probably mentioned this initially, but the reason I needed to use labels when performing the source reconstruction is because the recordings would not fit in memory.
I didn't need to have the source estimates be label-based in the first place, if not for the fact that my recordings are pretty long and I don't have a lot of RAM on my current computer.

Maybe if there was a way to use sparse matrices it would be possible, but due to how the setter is coded, I can't set `mne.SourceEstimate.data` to anything other than a numpy array.

MerlinDumeur, 2020-05-07:
I solved my problem by rewriting the inverse and morphing functions to stream from and to a hdf5 file instead of doing everything in memory.

closed 2020-05-07 by MerlinDumeur