mne-tools/mne-python #7732: Bug: event_repeated metadata incompatibility

kingjr, 2020-05-05:
`mne.Epochs` `event_repeated` parameter does not play well `metadata`:

```python
import mne
import numpy as np
import pandas as pd
raw_data = np.random.randn(2, 10000)
info = mne.create_info(2, 1000.)
raw = mne.io.RawArray(raw_data, info)

opts = dict(raw=raw, tmin=0, tmax=.001, baseline=None)
```

No problem when there are no repeated events and metadata
```python
events = [[0, 0, 1], [1, 0, 1]]
metadata = pd.DataFrame(events, columns=['onset', 'duration', 'value'])

epochs = mne.Epochs(events=events, metadata=metadata, **opts)
```

No problem when repeated events and no metadata
```python
events = [[1, 0, 1], [1, 0, 1]]
epochs = mne.Epochs(events=events, event_repeated='drop', **opts)
```

problem when repeated events and metadata
```python
events = [[1, 0, 1], [1, 0, 1]]
epochs = mne.Epochs(events=events, event_repeated='drop', metadata=metadata, **opts)
```

kingjr, 2020-05-05:
BTW, there should be an event_repeated='ignore' option. 

It is common to have distinct events types which end up with simultaneous onsets e.g. :
- during streamed stimulation,  button presses may be simultaneous to stim onset; 
- the beginning of words if always the beginning of a phoneme.

In such cases I don't think there is a good reason to force users to build different `Epochs` objects for each event type.

agramfort, 2020-05-05:
event_repeated='ignore' would lead to a duplicated epoch right?

if so no objection if it's well documented

>

kingjr, 2020-05-05:
duplicate in terms of m/eeg, but not in terms of metadata

agramfort, 2020-05-05:
so you expect the sample idx to be like an index in pandas sense in the
metadata?

kingjr, 2020-05-05:
not sure what you have in mind exactly; let me clarify my view

IMO the events and metadata should in principle be a unique object, i.e. a
sparse time series. Historically, they have been built separately but they
should remain connected during e.g. drop_bad, epoch indexing etc.

The ideal case would be for metadata  to be explicitly indexed by the event
onset (in raw sample, i.e. events[:,0]), but I *think* that would create
some backward incompatibilities.

The alternative is to link the events rows and the metadata iloc. One risk
I foresee with pd dataframes is the loc versus iloc indexing and the
reset_index mess, which could confuse users: dealing with pandas
indexing-related issues often require particular arguments to read and
write csv etc.

However, sparse time series / metadata are multidimensional (whereas events
is basically a unidimensional one hot encoder). We should thus allow
multiple events to start at the same time sample.

There are many cases where distinct events / dimensions of an event may
occur simultaneously: two distinct button presses in typing study, two
different words in an attention speech paradigm, during heavy decimation to
have a quick look etc. Dealing with the requirement in such cases is
painful, because it requires a non trivial find and arbitrary select
duplicate routine, which may be dependent on the subsequent analyses.

HTH

PS: no time for pr, apologies

agramfort, 2020-05-06:
> IMO the events and metadata should in principle be a unique object, i.e. a
> sparse time series. Historically, they have been built separately but they
> should remain connected during e.g. drop_bad, [...]

that's what I meant with pandas indexing and yes it's not an option
and not what we decided together some years ago

> The alternative is to link the events rows and the metadata iloc. One risk
> I foresee with pd dataframes is the loc versus iloc indexing and the
> reset_index mess, which could confuse users: dealing with pandas
> indexing-related issues often require particular arguments to read and
> write csv etc.
>

we decided that index is not used in metadata and that we rely on iloc

> However, sparse time series / metadata are multidimensional (whereas events
> is basically a unidimensional one hot encoder). We should thus allow
> multiple events to start at the same time sample.
>

I would just duplicate the rows in the metadata

>