mne-tools/mne-python #7738: raw.plot_psd: edge line of area plot

dnacombo, 2020-05-06:
when plotting raw psd with `n_fft=len(raw.times), average=True` to get high def in frequency, the edge of the grey shaded area overplots, which creates an ambiguous darker grey area around the normal grey area
[image]
Here's a zoomed in version:
[image]

would be nice to be able to disable this edge line in the call to `raw.plot_psd`

agramfort, 2020-05-06:
you can always edit a posteriori a matplotlib figure and remove the line.
Let me know if you need help with this.
is it just for a one time thing?

>

dnacombo, 2020-05-06:
yeh, I don't plan on doing these plots so often.
How do i find the right property in the matplotlib figure ?

agramfort, 2020-05-06:
this should do the job:

import matplotlib
import mne
from mne.datasets import sample

data_path = sample.data_path()

raw = mne.io.read_raw_fif(data_path + '/MEG/sample/sample_audvis_raw.fif')
raw = raw.pick_types(meg=False, eeg=True, eog=True, ecg=True, stim=True,
                     exclude=raw.info['bads']).load_data()

fig = raw.plot_psd()

lines = [ch for ch in ax.get_children() if isinstance(ch,
matplotlib.lines.Line2D)]
lines[0].set_visible(False)

>

hoechenberger, 2020-05-07:
IMHO this outer line should be disabled by default (or have the same color as the filled area). It doesn't add any additional value and leads to problems in edge cases like the one @dnacombo described. We could consider adding a kwarg to turn it on / set its color if users actually want to plot it…

agramfort, 2020-05-07:
this line tells you how data have been low passed

see: https://mne.tools/stable/auto_tutorials/raw/plot_40_visualize_raw.html#sphx-glr-auto-tutorials-raw-plot-40-visualize-raw-py

now I agree that grey on black when you don't use spatial_colors it's
not really a good choice

my 2c

larsoner, 2020-05-07:
@agramfort I think they're talking about the lines at the top and bottom of the shaded area, and you are talking about the dashed line to indicate the lowpass or line freq (probably line in this case?). 

I think I agree with @hoechenberger that having a line at the edges of the area does not actually add information. @hoechenberger you want to open a PR to remove it, and touch the example @agramfort linked to so we can see how it looks?

hoechenberger, 2020-05-07:
> I think I agree with @hoechenberger that having a line at the edges of the area does not actually add information. @hoechenberger you want to open a PR to remove it, and touch the example @agramfort linked to so we can see how it looks?

Yep can do!

PR #7750 (2020-05-07): Remove area edge lines in average PSD plots

#### Reference issue
Closes #7738

#### What does this implement/fix?
Do not plot lines on the edges of PSD areas
closed 2020-05-07 by larsoner