mne-tools/mne-python #7741: error in find_bad_channels_maxwell

SophieHerbst, 2020-05-06:
For one file out of many for which it works fine, I get an error in find_bad_channels_maxwell that I am not able to fix. 

> 
>  File "/home/sh254795/anaconda3/lib/python3.7/site-packages/mne/preprocessing/maxwell.py", line 1950, in find_bad_channels_maxwell
>     ss[-1] = end
> 
> IndexError: index -1 is out of bounds for axis 0 with size 0

Here is a minimal example:

import mne

filename = 'example_badchan_error-raw.fif'

raw = mne.io.read_raw_fif(filename,
                                  allow_maxshield=True,
                                  preload=True, verbose='error')

auto_noisy_chs, auto_flat_chs = mne.preprocessing.find_bad_channels_maxwell(raw,
             limit = 10., verbose=True, skip_by_annotation = 'BAD_ACQ_SKIP')

agramfort, 2020-05-06:
can you share the file so we can debug this?

>

larsoner, 2020-05-06:
I can see in the code where this can happen if there is some tiny segment of good data, will come up with a test and fix

larsoner, 2020-05-06:
... actually it looks like the file is shared above already, thanks @SophieHerbst !

SophieHerbst, 2020-05-06:
It does! awesome, thanks for the fast help @larsoner!

closed 2020-05-07 by larsoner

gabrielbmotta, 2020-07-24:
Does anyone have a link to the file mentioned above? We are trying to fix a similar issue in MNE CPP. Thanks.

larsoner, 2020-07-24:
@gabrielbmotta you shouldn't need the file above, you should just need to be on latest mne-python `master` dev branch. There is a fix in there already

gabrielbmotta, 2020-07-24:
@larsoner thanks for the quick response. We don't have that same function in our code, we're still trying to find out where and why a segfault happens when the file was loaded.

PR #7742 (2020-05-07): MRG, FIX: Fix exact match in find_bad_channels_maxwell
Closes #7741

@SophieHerbst can you see if it fixes your issue?