mne-tools/mne-python #7757: Accidentally pushed backport to upstream

cbrnr, 2020-05-09:
Oops, I accidentally pushed to the maint/0.20 branch in upstream instead of my fork. Sorry about that. Should I revert? It's only a few changes so maybe it's less effort if someone quickly reviews the commit (https://github.com/mne-tools/mne-python/commit/cb52773ea13bb94e533e372668bc536d5a2cdaf8)? In any case, this won't happen again (I don't know why I didn't track the branch on my fork).

agramfort, 2020-05-09:
yes it's fine to keep this commit. No harm done here.

>

cbrnr, 2020-05-09:
Would it be OK to make a new release sometime next week?

agramfort, 2020-05-09:
@larsoner the floor is yours

larsoner, 2020-05-09:
Yep, we can cut a 0.20.4, probably Monday

agramfort, 2020-05-10:
thanks @larsoner !

>

hoechenberger, 2020-05-11:
To prevent such things from happening, we should consider enabling protected branches. WDYT?

https://help.github.com/en/github/administering-a-repository/about-protected-branches

(I, for one, am totally in favor of making use of this feature.)

cbrnr, 2020-05-11:
You mean merging requires e.g. at least one positive review? Yes, why not - if it doesn't make other things more complicated.

hoechenberger, 2020-05-11:
There's quite a few options there, I _thought_ one could also disable direct pushs to certain branches; but maybe this is only possible indirectly by requiring a positive review. At any rate, I think it's worthwhile checking this out (no pun intended!)

cbrnr, 2020-05-11:
Yes, disabling direct pushes to upstream might be a good solution, because we should always go through our forks.

On a slightly related note, what about the https://github.com/mne-tools/mne-python/tree/old_max_power_ori branch in upstream? Shouldn't that also be located in a fork? @wmvanvliet

agramfort, 2020-05-11:
it means I won't be able to push hot fixes anymore :)

let's see how often issues occur before adding new rules

hoechenberger, 2020-05-11:
> it means I won't be able to push hot fixes anymore :)

Well. You shouldn't introduce bugs that require a hotfix in the first place! **Think about that!** 😉😅🙈

drammock, 2020-05-11:
> it means I won't be able to push hot fixes anymore :)

I think you could still push a hotfix to your fork and merge a hotfix PR yourself.  Probably we should *either* require an approving review *or* protect the branches, but not both (so that @agramfort can still get his hotfix fix)

cbrnr, 2020-05-11:
I'd vote for protected branches instead of review.

agramfort, 2020-05-11:
I vote for "this was necessary for the last 10 years" so let's not create rules
for every case otherwise you'll end up with the judicial system we have in
modern countries :)

larsoner, 2020-05-11:
> On a slightly related note, what about the https://github.com/mne-tools/mne-python/tree/old_max_power_ori branch in upstream? Shouldn't that also be located in a fork?

Indeed that shouldn't be there. I've moved it to my fork, @wmvanvliet feel free to copy it over to yours if need be. However right now @britta-wstnr and I are working on fixing all the orientation computations and such so I doubt we'll need it.

larsoner, 2020-05-11:
0.20.4 released

closed 2020-05-11 by larsoner

drammock, 2020-05-11:
while we're cleaning up old branches, `upstream/guidelines` is about 7000 commits behind `upstream/master` (diverged in 2015, but doesn't appear to have any unique commits).  OK if I kill it?

larsoner, 2020-05-11:
if it says "0 ahead" then yes please kill it, it doesn't contain anything we don't already have

cbrnr, 2020-05-11:
Thanks @larsoner!