mne-tools/mne-python #7765: mne.viz.plot_bem shows source space only in one slice

mariahakonen, 2020-05-11:
#### Describe the bug
I have created source space with mne.setup_volume_source_space and tried to visualize it with mne.viz.plot_bem. I have used sample data and ran the commands following this example: https://mne.tools/dev/auto_tutorials/source-modeling/plot_forward.html#sphx-glr-auto-tutorials-source-modeling-plot-forward-py

However, the source space is shown only in one slice.

I also got the same problem with my own data. 

#### Steps to reproduce
````
import os.path as op
import mne
from mne.datasets import sample
data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_raw.fif'

subjects_dir = data_path + '/subjects'
subject = 'sample'

surface = op.join(subjects_dir, subject, 'bem', 'inner_skull.surf')
vol_src = mne.setup_volume_source_space(subject, subjects_dir=subjects_dir,
                                        surface=surface)
print(vol_src)

mne.viz.plot_bem(subject=subject, subjects_dir=subjects_dir,
                 brain_surfaces='white', src=vol_src, orientation='coronal')

````
#### Expected results

[image]

#### Actual results

[image]

#### Additional information

Platform:      Linux-3.10.0-1062.4.3.el7.x86_64-x86_64-with-centos-7.8.2003-Core
Python:        3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 19:07:31)  [GCC 7.3.0]
Executable:    /share/apps2/anaconda/anaconda3/latest/envs/neuroimaging2/bin/python
CPU:           x86_64: 12 cores
Memory:        251.7 GB

mne:           0.19.1
numpy:         1.17.2 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.3.1
matplotlib:    3.1.1 {backend=Qt5Agg}

sklearn:       0.21.3
numba:         Not found
nibabel:       2.4.1
cupy:          Not found
pandas:        0.25.2
dipy:          Not found
mayavi:        4.6.2 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       Not found
vtk:           8.1.1

agramfort, 2020-05-11:
I cannot reproduce. I get the expected results with the code you shared.

can you try with the latest version of mne which is 0.20?

>

mariahakonen, 2020-05-12:
Thanks a lot! 
It seems to work with mne 0.20.4.

closed 2020-05-12 by mariahakonen