mne-tools/mne-python #7773: Parameter "reject" in ica.fit for epoch data object

langestroop, 2020-05-12:
Hi, 

this issue encompasses both MNE and autoreject. 

In [Autoreject Issue#170](https://github.com/autoreject/autoreject/issues/170) and in the MNE API page is reported that the reject parameter for [ica.fit](https://mne.tools/dev/generated/mne.preprocessing.ICA.html#mne.preprocessing.ICA.fit) can only be applied on a raw object.
However, in some cases, it would be benefitial do so directly on epochs, so @eioe and I were wondering whether there was a specific reason behind this choice? Do you think it would be possible to include the rejection parameter for epoch objects as well?

The question is mainly motivated by the fact that an ICA fitted on epoched data - that contain blinks/eye movements but where bad samples are ignored via the reject parameter - should provide a better decomposition compared to one fitted directly on continous "dirty" data. In our experience, removing intertrial intervals before ica fitting, drastically improves ica decomposition and leads to cleaner eye-components. It would be great to further improve upon this by also ignoring those of the remaining samples which are clearly artifactual (e.g., determined by a cut-off threshold).

larsoner, 2020-05-12:
> whether there was a specific reason behind this choice? Do you think it would be possible to include the rejection parameter for epoch objects as well?

You can already do [epochs.drop_bad(reject)](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.drop_bad), will that work for your use case?

langestroop, 2020-05-12:
Hi Eric, thanks for your speedy reply. 
This way, we would loose full epochs to single artifacts. Loosing many epochs is also problematic for the ica fitting. We were just wondering whether the same (sample based) procedure that is used for continuous data could also be applied within the epochs. This might leave us with epochs of different lengths, but for ica fitting this should be unproblematic, no? (We are projecting the weights to another copy of the data later).

agramfort, 2020-05-12:
to me also epochs.drop_bad(reject)
<https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.drop_bad> does
what you ask for

>

drammock, 2020-05-12:
> We were just wondering whether the same (sample based) procedure that is used for continuous data could also be applied within the epochs. This might leave us with epochs of different lengths

It is not possible to have epochs of different lengths, because of how the epoched data are stored internally.  Dropping a particular epoch is all-or-nothing, and can be achieved with `epochs.drop_bad()` as mentioned. By the way, the artifact rejection for `ICA.fit` on Raw data is not exactly "sample-based", how big a segment gets dropped depends on the value of the `tstep` parameter.

langestroop, 2020-05-14:
Hi all,

many thanks for your suggestions and insights: they're really useful. 
So basically a feasible approach would be something like this:

1. Epoch data
2. Compute autoreject global to find the proper rejection threshold
3. Apply rejection threshold via epochs.drop_bad to one copy of the dataset
4. Compute ICA on this dataset
5. Apply the weights to the other copy of the full dataset (it still contains bad epochs)
6. Only remove eyes/heart components 
7. Use autoreject local to actually take care of the bad epochs

Are we missing something?
If not, this would be great and it'd probably solves most of our issues. With the above mentioned pipeline we just see the danger that the threshold computed via autoreject global might also discard blinks together with otherwise dirty sections. Ideally we would need an algorithm that "masks" blinks while taking care of the dirt (similarly to what a human would manually do while exploring the dataset) to optimize the quality of the ICA (in terms of clean eye components) and to avoid to reject precious trials (just because of a blink) at the same time.
We realize that this is a more conceptual issue and goes beyond a simple fix feature request.The original question/issue is answered and can be closed from our side.

Thanks again for the insights and sorry for the lengthy post.

jasmainak, 2020-05-16:
Hi @langestroop sorry for the delayed response. Your pipeline looks reasonable to me. I would add that you can pass a `ch_types` parameter to `get_rejection_threshold` and leave out `eog` channel type. Secondly, I suspect that ICA would work just fine even if a few eyeblinks are missing due to `autoreject` global threshold since the eyeblink artifacts are quite stereotypical.

what can we do on `autoreject` end to make your life easier? should `get_rejection_threshold`  also accept a `raw` object? I'd be happy to make fixes / tutorials if you can cross-post your last comment on `autoreject` github so we don't lose it.

langestroop, 2020-05-26:
Hi @jasmainak,

sorry for the long delay and thanks for your reply and comments. 
Should I cross-post my previous comment directly in [Autoreject Issue#170](https://github.com/autoreject/autoreject/issues/170)? Or should I create a new issue in the autoreject github? Just let me know your preference and I'll do it right away.

Yes, I think that allowing `get_rejection_threshold` to accept a `raw` object would definitely be useful. In general, I'm currently thinking about other questions/ideas for fixes/tutorials. Many thanks again for your availability.

jasmainak, 2020-05-27:
I think cross posting as a new issue with a summary of what changes in tutorials etc can help would be very welcome! Better still if you want to make the PR yourself, but I won't ask for more than you have time for :)