mne-tools/mne-python #7778: detrend existing epochs?

SophieHerbst, 2020-05-13:
I cannot find out how to apply detrending to allready created epochs. Is there a function similar to 
[evoked.detrend](https://mne.tools/stable/generated/mne.Evoked.html?highlight=detrend#mne.Evoked.detrend)?

I know I can detrend when creating epochs, but I would like to create un-detrended epochs and just detrend for a specifc purpose. 

Given that it is mentioned in the [epochs doc](https://mne.tools/stable/generated/mne.Epochs.html), it should be easy to apply later (says @hoechenberger)?

> When accessing data, Epochs are detrended, baseline-corrected, and decimated, then projectors are (optionally) applied.

hoechenberger, 2020-05-13:
Can you try setting `epochs.detrend` to the desired value? This should do what you want!

I.e., if you created them with `epochs = Epochs(..., detrend=None`, you should later be able to do `epochs.detrend = 1`, and accessing the data should give you the correct results – at least if I'm understanding the code correctly.

SophieHerbst, 2020-05-13:
What do you mean by 'accessing the data'? I tried epochs.plot(), epochs.apply_proj() and epochs.load_data() after setting detrend to 1, but I don't see any change. 

It seems to be applied when set before loading the data: 
```
epochs = mne.read_epochs(fname_in, preload=False)
epochs.detrend=1
epochs.plot(scalings={'grad': 2000e-13, 'mag': 2000e-15},n_epochs=10) 
```

Is there away to get it applied to already loaded data?

hoechenberger, 2020-05-13:
Can you try:

```
epochs = mne.read_epochs(fname_in, detrend=None)
epochs.plot(scalings={'grad': 2000e-13, 'mag': 2000e-15},n_epochs=10) 

# Set detrend flag and reload the data to apply it.
epochs.detrend = 1
epochs.preload = False
epochs.load_data()
epochs.plot(scalings={'grad': 2000e-13, 'mag': 2000e-15},n_epochs=10) 
```

hoechenberger, 2020-05-13:
@SophieHerbst Whoops apparently you have to reset the `Raw` object reference inside `epochs` too… So this works for me:

```python
import mne
from mne import io
from mne.datasets import sample

data_path = sample.data_path()

raw_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw.fif'
event_fname = data_path + '/MEG/sample/sample_audvis_filt-0-40_raw-eve.fif'
event_id, tmin, tmax = 1, -0.2, 0.5

raw = io.read_raw_fif(raw_fname)
events = mne.read_events(event_fname)

epochs = mne.Epochs(raw, events, event_id, tmin, tmax,
                    baseline=(None, 0), preload=True)
epochs.plot()

epochs_detrended = epochs.copy()
epochs_detrended.detrend = 1
epochs_detrended.preload = False
epochs_detrended._raw = raw.copy()
epochs_detrended.load_data()
epochs_detrended.plot()

# Check that something has changed
import numpy as np
assert not np.allclose(epochs.get_data(),
                       epochs_detrended.get_data())
```
But this is kind of awkward, because you would have to keep the `raw` around :(

hoechenberger, 2020-05-13:
Workaround:
```python
epochs_not_preloaded = mne.Epochs(raw, events, event_id, tmin, tmax,
                                  baseline=(None, 0), preload=False)

epochs= epochs_not_preloaded.copy().load_data()

epochs_detrended = epochs_not_preloaded.copy()
epochs_detrended.detrend = 1
epochs_detrended.load_data()
```

SophieHerbst, 2020-05-13:
mh. so in conclusions, already loaded epochs can not be detrended. Would it be difficult to add that function, given it exists for evokeds?

hoechenberger, 2020-05-13:
So currently, loading the data  removes the reference to the raw object the epochs were built from. If we don’t do this, we could easily detrend at any later time. 

@agramfort @larsoner any idea why the reference isn’t kept? Memory freeing?

larsoner, 2020-05-13:
> any idea why the reference isn’t kept? Memory freeing?

Yes

> Would it be difficult to add that function, given it exists for evokeds?

Yes we could add a method to detrend already loaded data, just like we have for filtering

hoechenberger, 2020-05-13:
> Yes we could add a method to detrend already loaded data, just like we have for filtering

It wouldn't work with `Epochs` loaded from disk though, or would it?

larsoner, 2020-05-13:
Yes the idea is that `epochs.detrend` would only work on epochs that have been loaded into memory / from disk already

hoechenberger, 2020-05-13:
> Yes the idea is that `epochs.detrend_data()` (or whatever we call it) would only work on epochs that have been loaded into memory / from disk already

Ok got it. I was confused because I thought that, for whatever reason, we need access to the `Raw` to perform detrending. Looking at the code again, this seems not to be the case. :) Nice!

kingjr, 2020-09-18:
in the meantime, a quick fix is
```python
epochs.data[:] = scipy.signal.detrend(epochs.get_data(), axis=-1, fit='linear')
```

larsoner, 2023-10-01:
I think a reasonable way to get this behavior is [`epochs.apply_function`](https://mne.tools/stable/generated/mne.Epochs.html#mne.Epochs.apply_function) so I think we can close this

closed 2023-10-01 by larsoner

3x10e8, 2024-03-02:
For completeness, I would just mention the function calls we could use for DC vs. linear detrending:

For detrend = 0 behavior:
`epochs.apply_function(scipy.signal.detrend, type='constant')`

And for detrend = 1:
`epochs.apply_function(scipy.signal.detrend, type='linear') # default is linear for scipy's detrend`