mne-tools/mne-python #7779: ENH: make out_type default to 'indices' for permutation tests

yh-luo, 2020-05-13:
#### Describe the problem
Permutation tests crashed when datasets are large (50 time points, oct6, 13 subjects  in my case). `mne.stats.permutation_cluster_1samp_test` with `out_type='mask'` caused memory overflow and the process was killed on a machine with 64GB RAM and 16GB swap. With `out_type='indices'`, the process only uses about 4GB RAM.
Since `out_type` is default to `'indices'` for `mne.stats.spatio_temporal_cluster_1samp_test` and `mne.stats.spatio_temporal_cluster_test`, it would be nice that `mne.stats.permutation_cluster_1samp_test`, `mne.stats.permutation_cluster_test` also default to `'indices'` to avoid the memory error.

#### Describe your solution
Current:
```mne.stats.permutation_cluster_1samp_test(..., out_type='mask', ...)```

Proposed:
```mne.stats.permutation_cluster_1samp_test(..., out_type='indices', ...)```

drammock, 2020-05-13:
+1 for this proposal. Another reason: just yesterday I noticed that the default results of `permutation_cluster_test` are not compatible with `summarize_clusters_stc` because it requires `out_type='indices'` in order to work.

larsoner, 2020-05-13:
I can live with a deprecation cycle to get `out_type` to be consistent across these functions

agramfort, 2020-05-13:
agreed

>

commit f8276a8c (2020-05-13): MRG, API: change out_type default in permutation_cluster_(1samp_)test (#7781)

* change out_type default; closes #7779

* missed one

* flake whitespace

* more whitespace

* update what's new

* fix docstring
closed 2020-05-13 by drammock