mne-tools/mne-python #7782: DOC: old versions of docs website should link to current stable

drammock, 2020-05-13:
Lots of packages (numpy, scipy, etc) have prominent messages on their outdated docs websites, saying something like

> You are viewing docs for version XX, click here for current version (YY) docs

We need to do this too; a colleague of mine just got routed by google to the docs from version 0.15 (!) and didn't immediately notice the outdated version.

Additionally, I've mentioned before somewhere that the old docs version dropdown doesn't show versions newer than itself:

[image]

But that's a more annoying problem to fix than just adding a big colorful banner at the top saying "these docs are old", and I think if we have the big banner then we can ignore the dropdown problem.

agramfort, 2020-05-13:
that sounds painful to fix ...

>

larsoner, 2020-05-13:
It could probably be done automatically by constructing an appropriate regular expression substitution. A nice Sunday hack for someone who wants to expand their regex horizons...

drammock, 2020-05-13:
since numpy, sklearn etc already do this, we can just see how they do it and copy shamelessly.

larsoner, 2020-05-13:
They probably use a systematic automated versioning system. We use the few commands in the Wiki. Would probably be a lot of work to update to a new system. We could undertake it I guess. But I would be inclined just to do the regex fix for now, and make that part of the release process somehow (it's not bad to do those commands every 6 months).

hoechenberger, 2020-05-17:
I’m regularly running into this issue too. I google for some MNE command or class and sometimes I end up on some VERY outdated documentation. In fact, it’s typically the exception that the docs of the most recent stable version are the first hit

agramfort, 2020-05-18:
yes it would be nice to fix

>

drammock, 2020-05-19:
> They probably use a systematic automated versioning system. We use the few commands in the Wiki. Would probably be a lot of work to update to a new system.

Here is what sklearn uses: view-source:https://scikit-learn.org/versionwarning.js

I think we could adapt it to the specifics of our site, and it would only add 1 task to the release workflow on the wiki (updating the line `var latestStable = '0.20.5';` at the top of the script).  There would be a one-time cost of hacking a call to this script into all of the old doc website pages, but henceforth it could be made part of the template.

larsoner, 2020-05-19:
Sounds good to me

hoechenberger, 2020-05-19:
Great find, @drammock!! I would love to see this implemented :)

larsoner, 2020-05-20:
FYI @drammock this landed in SciPy in https://github.com/scipy/scipy/pull/12168 and https://github.com/scipy/docs.scipy.org/pull/38

drammock, 2020-05-20:
- [x] adapt JS [mne-tools.github.io#19](https://github.com/mne-tools/mne-tools.github.io/pull/19)
- [ ] update styling
- [ ] test on Circle

PR #7804 (2020-05-21): MRG: warning banner for old versions of docs
closes #7782 

- adds a call to `versionwarning.js` to the layout template
- styles the links in the resulting devbar

WIP until we iron out the style-related questions in [mne-tools.github.io#20](https://github.com/mne-tools/mne-tools.github.io/pull/20)
closed 2020-05-21 by larsoner