mne-tools/mne-python #7788: remove spyder from environment file?

drammock, 2020-05-14:
Seems like every 6-12 months, one of my colleagues says "Spyder won't load, and in trying to fix it I borked my environment, please help."  I'm wondering why we implicitly recommend spyder by including it in our `environment.yml` file.  Spyder also imposes version constraints (e.g., currently it requires `PyQt5<5.13`), which has the potential to cause conflicts / inconsistent environments if (for example) pyvista and spyder impose incompatible dependency constraints.

Shouldn't we leave it up to the end user to pick their IDE?  If not, can we be a bit more intentional about our recommendations?

larsoner, 2020-05-14:
Okay with me

larsoner, 2020-05-14:
(to remove it from the environment file I mean)

agramfort, 2020-05-15:
agreed

hoechenberger, 2020-05-17:
Should be dropped.

PR #7796 (2020-05-19): MRG, MAINT, DOC: Remove spyder
closes #7788 

- removes spyder from our environment file
- adds docs section on installing IDEs
- small clarification to pre-install docs
closed 2020-05-19 by agramfort