mne-tools/mne-python #7789: I've a problem with "plot_sensors_connectivity" visualization

Sirabhop, 2020-05-15:
My goal is to compute EEG connectivity, so I run this code below in order to compute connectivity.
`con, freqs, times, n_epochs, n_tapers = spectral_connectivity(epoch, method='pli', mode='multitaper', sfreq=2048, fmin=4, fmax=8, faverage=True, mt_adaptive=False, n_jobs=1)`

And, I also **need to visualize the connectivity**, so I use this code
`plot_sensors_connectivity(epoch.info, con[:, :, 0])`
But, turn out it gave me this error

> AttributeError: 'NoneType' object has no attribute '_Renderer'

(I have used EEG biosemi32 measuring brain wave, set montage with 'biosemi32')
`epoch.set_montage('biosemi32')`

Thank you

GuillaumeFavelier, 2020-05-15:
Hello @Sirabhop and thank you for your reporting your issue.

Please could you share the output of:

```
python -c "import mne; mne.sys_info()"
```

Plus maybe a minimal example to reproduce? It would be really helpful.

Sirabhop, 2020-05-15:
`mne.sys_info()`

> Platform:      Linux-4.19.104+-x86_64-with-Ubuntu-18.04-bionic
> Python:        3.6.9 (default, Apr 18 2020, 01:56:04)  [GCC 8.4.0]
> Executable:    /usr/bin/python3
> CPU:           x86_64: 2 cores
> Memory:        12.7 GB
> 
> mne:           0.20.4
> numpy:         1.18.4 {blas=openblas, lapack=openblas}
> scipy:         1.4.1
> matplotlib:    3.2.1 {backend=module://ipykernel.pylab.backend_inline}
> 
> sklearn:       0.22.2.post1
> numba:         0.48.0
> nibabel:       3.0.2
> cupy:          Not found
> pandas:        1.0.3
> dipy:          Not found
> mayavi:        Not found
> pyvista:       0.24.2
> vtk:           8.1.2

GuillaumeFavelier, 2020-05-15:
I see that you have `pyvista` installed and not `mayavi`. Could you also try:

```
python -c "import pyvista; print(pyvista.Report())"
```

Sirabhop, 2020-05-15:
I got session crashed

> ERROR:root:bad X server connection. DISPLAY=

Is it about my computer? because I run all my code via Google Colab

GuillaumeFavelier, 2020-05-15:
> ERROR:root:bad X server connection. DISPLAY=

AFAIK, this means that the environment does not provide a window with an opengl context. I'm afraid this is an implicit requirement for now with the `pyvista` 3d backend (it's more a `vtk` issue really).

A way to mitigate could be to use a `xvfb` server to simulate this but I'm not familiar enough with Google Colab to help you set up that. A config example is available in [.circleci/config](https://github.com/mne-tools/mne-python/blob/master/.circleci/config.yml).

On the bright side, there is active work to add support for headless remote visualization as we speak in https://github.com/mne-tools/mne-python/pull/7758 for example.

Any other ideas to make it work @agramfort , @hoechenberger ?

GuillaumeFavelier, 2020-06-29:
Little update on this. With #7758 merged, there is now a "notebook" 3d backend available! This has been tested to work as expected on Jupyter Notebook.

About support for Google Colab, I tried setting it up with @drammock without much success so far. Mainly because we require `mesalib` from `conda` and configuring `conda` is not straightforward in this scenario...

FYI, I plan to move on to Jupyterlab support next.