mne-tools/mne-python #7790: Error in reading data from Fieldtrip

gbrookshire, 2020-05-15:
#### Describe the bug

I'm unable to read data saved by Fieldtrip using the functions in `mne.io.fieldtrip`. Any suggestions would be greatly appreciated!

#### Steps to reproduce

Download the Fieldtrip data from this link:
<ftp://ftp.fieldtriptoolbox.org/pub/fieldtrip/tutorial/coherence/data.mat>

```python
import mne
data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', None)
```

#### Expected results
Load the data saved in the Fieldtrip object.

#### Actual results
```
In [5]: data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', None)                                
<ipython-input-5-6596278ed022>:1: RuntimeWarning: Importing FieldTrip data without an info dict from the original file. Channel locations, orientations and types will be incorrect. The imported data cannot be used for source analysis, channel interpolation etc.
  data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', None)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-5-6596278ed022> in <module>
----> 1 data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', None)

~/anaconda3/envs/mne/lib/python3.7/site-packages/mne/io/fieldtrip/fieldtrip.py in read_epochs_fieldtrip(fname, info, data_name, trialinfo_column)
    117     ft_struct = ft_struct[data_name]
    118 
--> 119     info = _create_info(ft_struct, info)  # create info structure
    120     data = np.array(ft_struct['trial'])  # create the epochs data array
    121     events = _create_events(ft_struct, trialinfo_column)

~/anaconda3/envs/mne/lib/python3.7/site-packages/mne/io/fieldtrip/utils.py in _create_info(ft_struct, raw_info)
     60         pick_info(info, ch_idx, copy=False)
     61     else:
---> 62         montage = _create_montage(ft_struct)
     63 
     64         info = create_info(ch_names, sfreq, montage=montage)

~/anaconda3/envs/mne/lib/python3.7/site-packages/mne/io/fieldtrip/utils.py in _create_montage(ft_struct)
    143                 cur_labels[available_channels])
    144             montage_pos.extend(
--> 145                 cur_ch_struct['chanpos'][available_channels])
    146 
    147     montage = None

KeyError: 'chanpos'
```

#### Additional information

The data loads correctly using Matlab.
```
>> load ../data/data.mat
>> data

data = 

  struct with fields:

        hdr: [1×1 struct]
      label: {153×1 cell}
      trial: {1×164 cell}
       time: {1×164 cell}
    fsample: 1200
       grad: [1×1 struct]
        cfg: [1×1 struct]
```

I showed this error using the stable release of `mne`, but I get the same result with the current version on Github.

```
mne.sys_info()                                                                                                 
Platform:      Linux-4.15.0-96-generic-x86_64-with-debian-buster-sid
Python:        3.7.6 (default, Jan  8 2020, 19:59:22)  [GCC 7.3.0]
Executable:   .../anaconda3/envs/mne/bin/python
CPU:           x86_64: 4 cores
Memory:        39.1 GB

mne:           0.19.1
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=Qt5Agg}

sklearn:       0.22.1
numba:         0.48.0
nibabel:       3.0.2
cupy:          Not found
pandas:        1.0.3
dipy:          1.1.1
mayavi:        4.7.1 {qt_api=pyqt5, PyQt5=5.9.2}
pyvista:       0.24.0
vtk:           8.2.0
```

gbrookshire, 2020-05-15:
Reading the data with the `info` object read in from the raw data leads to a different error.

Original raw data is available here:
ftp://ftp.fieldtriptoolbox.org/pub/fieldtrip/tutorial/SubjectCMC.zip

```python 
raw_ctf = mne.io.read_raw_ctf('SubjectCMC/SubjectCMC.ds')
original_info = raw_ctf.info
data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', original_info)
```

The info seems to show up, although the channel names have extra numbers attached to the end:
```
In [12]: original_info                                                                                                 
Out[12]: 
<Info | 21 non-empty fields
    bads : list | 0 items
    ch_names : list | STIM, SCLK01-177, BG1-1706, BG2-1706, BG3-1706, ...
    chs : list | 191 items (STIM: 1, MISC: 3, REF_MEG: 32, MAG: 151, EEG: 4)
    comps : list | 5 items
    ctf_head_t : Transform | 3 items
    custom_ref_applied : bool | False
    dev_ctf_t : Transform | 3 items
    dev_head_t : Transform | 3 items
    dig : Digitization | 7 items (3 Cardinal, 4 EEG)
    events : list | 0 items
    highpass : float | 0.0 Hz
    hpi_meas : list | 0 items
    hpi_results : list | 1 items
    lowpass : float | 600.0 Hz
    meas_date : NoneType | unspecified
    meas_id : dict | 4 items
    nchan : int | 191
    proc_history : list | 0 items
    projs : list | 0 items
    sfreq : float | 1200.0 Hz
    subject_info : dict | 1 items
    acq_pars : NoneType
    acq_stim : NoneType
    description : NoneType
    device_info : NoneType
    experimenter : NoneType
    file_id : NoneType
    gantry_angle : NoneType
    helium_info : NoneType
    hpi_subsystem : NoneType
    kit_system_id : NoneType
    line_freq : NoneType
    proj_id : NoneType
    proj_name : NoneType
    utc_offset : NoneType
    xplotter_layout : NoneType
>
```

Here's the new error when trying to read the data:
```
In [13]: data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', original_info)                              
<ipython-input-13-a19090e3ca3a>:1: RuntimeWarning: The following channels are present in the FieldTrip data but cannot be found in the provided info: {'MLT24', 'MLT34', 'MLT31', 'MLC13', 'MRF34', 'MLO41', 'MRF21', 'MRF51', 'MLT12', 'MRT13', 'MRT43', 'MRC32', 'MLF21', 'MRC11', 'MRC41', 'MRC42', 'MLC24', 'MLF41', 'MRF31', 'MRT41', 'MLP12', 'MRT14', 'MRO21', 'MLF31', 'MZP02', 'MRT22', 'MRF43', 'MLF44', 'MRP32', 'MLP21', 'MRP33', 'MRT42', 'MRC21', 'MLF22', 'MLT32', 'MLC12', 'MLC41', 'MLT15', 'MRT44', 'MZF02', 'MLC22', 'MLT42', 'MRO11', 'MRF44', 'MLT23', 'MLC32', 'MLF34', 'MLF42', 'MLC11', 'MLP32', 'MLT16', 'MRO22', 'MRO31', 'MRF41', 'MLC42', 'MRT16', 'MZO02', 'MRT33', 'MLT44', 'MZC02', 'MRC33', 'MZC01', 'MRF52', 'MZF03', 'MRT25', 'MLT13', 'MRF42', 'MRT26', 'MRO12', 'MLP33', 'MRC14', 'MLF33', 'MLO42', 'MRF32', 'MRO33', 'MRT24', 'MLF43', 'MRT34', 'MRT21', 'MRF22', 'MRC13', 'MRT31', 'MLF23', 'MLT22', 'MLP11', 'MLT41', 'MLT14', 'MLT26', 'MLC14', 'MLO43', 'MLO12', 'MZF01', 'MRO41', 'MRC22', 'MLO22', 'MLF52', 'MLO31', 'MRT11', 'MRC31', 'MRF45', 'MLO32', 'MLF12', 'MRF11', 'MLC43', 'MLT35', 'MRO42', 'MLC33', 'MRT35', 'MLT25', 'MLT33', 'MLP13', 'MRP31', 'MLF45', 'MLT21', 'MLF11', 'MRP13', 'MRP12', 'MLC23', 'MLO21', 'MZP01', 'MLO33', 'MLP34', 'MRC15', 'MLP31', 'MLF32', 'MRC12', 'MRF33', 'MRP34', 'MRC43', 'MRC23', 'MRO43', 'MRP11', 'MRF23', 'MLC31', 'MRO32', 'MRT15', 'MLP22', 'MLC21', 'MLT43', 'MLO11', 'MLF51', 'MLT11', 'MRC24', 'MRT23', 'MRT32', 'MZO01', 'MLC15', 'MRP21', 'MRP22', 'MRF12', 'MRT12'}.
These channels will be removed from the resulting data!
  data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', original_info)
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-13-a19090e3ca3a> in <module>
----> 1 data_from_ft = mne.io.fieldtrip.read_epochs_fieldtrip('data.mat', original_info)

~/anaconda3/envs/mne/lib/python3.7/site-packages/mne/io/fieldtrip/fieldtrip.py in read_epochs_fieldtrip(fname, info, data_name, trialinfo_column)
    117     ft_struct = ft_struct[data_name]
    118 
--> 119     info = _create_info(ft_struct, info)  # create info structure
    120     data = np.array(ft_struct['trial'])  # create the epochs data array
    121     events = _create_events(ft_struct, trialinfo_column)

~/anaconda3/envs/mne/lib/python3.7/site-packages/mne/io/fieldtrip/utils.py in _create_info(ft_struct, raw_info)
     51             ch_names = new_chs
     52             ft_struct['label'] = ch_names
---> 53             if ft_struct['trial'].ndim == 2:
     54                 ft_struct['trial'] = np.delete(ft_struct['trial'],
     55                                                missing_chan_idx,

AttributeError: 'list' object has no attribute 'ndim'
```

agramfort, 2020-05-15:
ping @thht

thht, 2020-05-15:
ok, i am going to take a look at it as soon as possible...

thht, 2020-05-20:
ok, i took a look at what is going on here. in fact, there were a few issues that you uncovered, @gbrookshire that are taken care of in #7802 . however, it won't be possible to read that datafile into MNE because the trials do not have the same time. this is supported in FieldTrip but not in MNE (as far as I am aware...). So importing this file will still raise an error, yet a more informative one.

thht, 2020-05-20:
@agramfort i think we can close this issue now as #7802 has been merged.

closed 2020-05-20 by larsoner

gbrookshire, 2020-05-27:
Great - thanks for your help with this!