mne-tools/mne-python #7793: Facilitate export to XArray

christian-oreilly, 2020-05-17:
#### Describe the problem
XArray is very convenient when doing data exploration as it often allows for a more transparent code by using dimension names rather than indexes. It would be very useful if MNE was providing XArray export functions where it already provides exports to NumPy array. For example, whereas someone would currently use 
```
ica.get_sources(epochs).get_data().mean(2).mean(0)
```
by adding a to_xarray() method, we could use the following alternative code
```
ica.get_sources(epochs).to_xarray().mean(["time", "epochs"])
#  or ica.get_sources(epochs).to_xarray().mean("time").mean("epochs")
```
Personally, I found the second version much easier to understand. With the first version, I would typically have to print the shape of the data to infer what axis corresponds to what to understand what operation is being performed. The first form (with dimension indexes) is also much more error-prone because all sorts of things can affect the axes ordering whereas dimension names and coordinates generally remain consistent and generate much more transparent errors when they don't.

agramfort, 2020-05-17:
no objection. We have to_data_frame so we can add a to_xarray following the same approach (using nested import)

larsoner, 2020-05-17:
Or maybe add a kwarg to get_data, output='ndarray' (default) or 'xarray'. I think I'd prefer that to a new function

agramfort, 2020-05-17:
it's just harder to discover

>

larsoner, 2020-05-17:
Indeed, but less API / kwarg duplication

hoechenberger, 2020-05-17:
I’d prefer dedicated functions for each export. It’s how all other packages behave that I’ve ever encountered in my scientific Python stack.

christian-oreilly, 2022-09-18:
Any objections to adding xarray in the requirements.txt? I am working on a different PR related to this https://mne.discourse.group/t/exporting-surface-sources-as-nifti-volume/5518/3 I'd used xarray in that PR, but that would need adding this dependency. I am thinking if we were to implement this feature request (I can do a PR for this), it would require adding this dependency anyway. I think XArray is a great component of Python scientific stack and worth integrating. I want to check first as I understand that there are also valid arguments for keeping dependencies as light as possible.

larsoner, 2022-09-18:
I don't think you should need `xarray` to convert a surface to NIfTI at least

christian-oreilly, 2022-09-18:
@agramfort  Yes, I agree that adding dependencies must not be a rash decision. Some thoughts on that:
1 - The XArray export functionality would require importing XArray. You were thinking of it more as an extra/optional import required and checked for only when calling such an export function?
2 - I do use it for the intended PR for things like that:
```python
    source_data = xr.DataArray(stc.data, dims=["sources", "time"]) \
                    .rolling(time=time_downsample_factor, center=True) \
                    .mean()[:, time_downsample_factor//2:-time_downsample_factor//2:time_downsample_factor]
```
Of course, there is nothing there that cannot be implemented in plain Numpy or with Pandas (since in this specific context, I am only in 2D), but I think XArray significantly adds to readability as it allows operating on axes not using integers but with named dimensions. Compares 
```python
    source_data.transpose("time", "sources")
    source_data.sel(time=source_data.time > 0).mean("time")
```
with
```python
    source_data.transpose(1, 0):
    source_data[:, time > 0].mean(1)
    # Note also that here, time needs to be maintained
    # as a separate variable. XArray can maintain it internally
    # as the coordinates for the given dimension.
```
for example. The former is explicit, whereas the latter requires introspecting the objects to understand what is what. As much as Pandas is a game changer (compared to using Numpy only) when doing data analysis in general, XArray is a game changer when working with multidimensional data that can be represented as dense blocks (e.g., time X channel X epochs X ...); less so if not dense like if stacking data across subjects (e.g., time X channel X epoch X subject) where there is a large variability in the numbers of epochs per subject.
3 - XArray is an open door for using Dask (distributed computing) to scale up analyses toward "bigger" data applications.
4 - In the long term, I think mne could benefit from using XArray for some of its internal operations (to make them simpler and more readable).

Note that I have no particular involvement with XArray (i.e., I have no conflicting interests). I am just a happy customer :) 

@larsoner Agreed. It is not a blocker for converting a surface to NIfTI. If available, I'll use it. If not, I'll get around. So this does not need to be settled for this specific PR.

mmagnuski, 2023-02-09:
Over the last 1.5 year I've been using xarray very frequently and it has grown indispensible to me. It makes working with arrays just a smoother and more elegant experience. For example assigning the metadata as dimension coordinates and then performing groupby operations in calculations or plotting is very handy. I'd be happy to invest some time to bring `.to_xarray()` to mne objects (with relevant examples etc.). I'm wondering which object it would be a good idea to start with: maybe Epochs or Evoked?
also related issue (at least the example I posted there): #11319

larsoner, 2023-02-09:
@mmagnuski whichever supports your use case :)

I image this would only be a few lines of code, so the support burden would be low. It's an optional dependency so probably not so bad.

A good first step would actually be to open a WIP PR that just adds `xarray` to our `requirements.txt`, `environment.yml`, and `tools/*_dependencies.sh` for the `--pre` runs. That way we can see that at least dependency resolution isn't problematic. Then you could add the actual code with CI infrastructure already in place