mne-tools/mne-python #7798: Rename temporal_derivative_distribution_repair to tddr

cbrnr, 2020-05-19:
Can we rename `temporal_derivative_distribution_repair` to just `tddr`? I really think that the function name is too long (39 characters) - it makes it impossible to stay within the 79 character limit without using ugly `\` line breaks, e.g.:

```python
from mne.preprocessing.nirs import temporal_derivative_distribution_repair as tddr  # 82 characters
```

I think renaming to `tddr` is not problematic because the function is in the `nirs` subpackage, and TDDR is a commonly known abbreviation in the NIRS community (googling for "nirs tddr" immediately brings up several results with the correct method).

This issue came up in https://github.com/cbrnr/mnelab/pull/145 that @rob-luke is working on. He'd probably be fine with renaming because that's what he initially proposed.

And while we're at it, I'd like to rename `beer_lambert_law` to `beer_lambert` as well (we'd save 4 chars without losing any meaning).

rob-luke, 2020-05-19:
As a newbie to the python/MNE way I defer to your collective judgement. Im happy to implement whatever the group decides here.

agramfort, 2020-05-19:
I pass here. Honestly I think we should use editors that autocomplete so
typing
long function names is not an issue. I am more worried about make the code
self explanatory. I am not a NIRS expert nor are new comers in the field and
abbreviations and jargon are always an issue. Yes when you know it's obvious
and google certainly has the answer. We use ICA and not
independent_component_analysis
so I guess my argument has some limits but I prefer to be too
verbose/explicit than
implicit.

rob-luke, 2020-05-19:
Ok Im going to remove myself from the fence and vote for the longer more explicit naming.

cbrnr, 2020-05-19:
I vote for the shorter name.

cbrnr, 2020-05-19:
In general I agree with @agramfort, but this name is taking it too far. Plus it's like ICA, NIRS people should have heard of TDDR.

hoechenberger, 2020-05-19:
>  it's like ICA, NIRS people should have heard of TDDR.

Yes this seems to be true. I would, therefore, also appreciate the shorter name in this particular case.

cbrnr, 2020-05-19:
In case the majority is in favor of renaming I've already implemented it in #7800.

drammock, 2020-05-19:
is `temporal_deriv_distrib_repair` an acceptable compromise?  I think it keeps the abbreviated words long enough that they aren't easily mistaken for something completely different (e.g., shortening to "dist" could be "distribution" or "distance"...)

cbrnr, 2020-05-19:
If someone doesn't know the algorithm, the long name is not better in conveying what the function actually does. For me it wasn't obvious at all that this algorithm corrects motion artifacts - temporal derivation distribution repair is a very technical description of the algorithm but doesn't give many hints on its purpose. Therefore, I'd still vote for the short name because the compromise is neither short nor self-explanatory.

BTW, we do have plenty of abbreviations in MNE function names (`simulate_stc`, `ICA`, `compute_source_psd`, `CSP`, `SPoC`, `make_lcmv`, ...). I will create PRs and change these to their full names 😝 - just kidding. I think these names are perfectly OK, because if I don't know what they do I just read the docstrings.

larsoner, 2020-05-19:
@rob-luke are you convinced? I've come around to preferring +0.5 the short name as well...

hoechenberger, 2020-05-19:
> BTW, we do have plenty of abbreviations in MNE function names (`simulate_stc`, `ICA`, `compute_source_psd`, `CSP`, `SPoC`, `make_lcmv`, ...).

You forgot about … `set_electroencephalogram_reference` :)

cbrnr, 2020-05-19:
> You forgot about … set_electroencephalogram_reference :)

Hehe LOL - this was going to be my final threat 🤣.

agramfort, 2020-05-19:
KO :)

>

rob-luke, 2020-05-19:
My vote is still for the longer name. I don’t use abbreviations in journal articles any more either, as it makes it less accessible to newcomers to the field. I think the same applies here. Also you can just add ‘import x as y’ if concerned about character counts.

I don’t think this is comparable to acronyms like eeg. These are globally known (I wouldn’t write electroencephalogram throughout an article either). Domain specific acronyms make my head spin, there’s hundreds of them and every new paper invents one. However, I agree this one is extra long and technically, not functionally, descriptive.

rob-luke, 2020-05-19:
I have to say I’m quite happy people are discussing variable names I submitted, and not complaining about code bugs :)

larsoner, 2020-05-19:
Maybe this would be a good time to just do:
```
tddr = temporal_derivative_distribution_repair
```
in the module. Then people can choose which to use, and in the docstring we can make it clear that it's an alias. I don't think it's too bad -- no real extra maintenance burden, and the docstring can make it clear that there are aliases in the namespace.

I don't like it in general because it could make it hard to know what is what, but in this case it's pretty clear. The big problem to me with the longer name is as @cbrnr mentioned in the top is that you can't even PEP8-compliantly one-line the shorter name, it requires 2 lines. So maybe in cases where names are *this* long, we should provide convenience aliases in the same namespace? (Hopefully this will be the only case...)

> I have to say I’m quite happy people are discussing variable names I submitted, and not complaining about code bugs :)

Indeed I'm confident that this will be the most challenging bug we encounter with it :)

hoechenberger, 2020-05-19:
> I don’t use abbreviations in journal articles any more either, as it makes it less accessible to newcomers to the field. I think the same applies here.

In articles, I introduce all abbreviations, including common ones like EEG. Once introduced, I use the abbreviation only :)

> Also you can just add ‘import x as y’ if concerned about character counts.

Yes but then one always has to figure out the `as y` part every time one looks at user code, as long as their is no agreed-upon standard import (like `import numpy as np` or `import pandas as pd`)

> Domain specific acronyms make my head spin, there’s hundreds of them and every new paper invents one.

That's the point. They're, kind of, all domain-specific these days. And if your research covers this domain, you will get to know the acronym. If your research takes place in other domains, you can ignore the acronym, as you will not apply that method anyway…

rob-luke, 2020-05-19:
Yes I understand your arguments too. I think each of us weighs the different pros and cons differently. Engineers and coders love acronyms ;) .I’ve cast my vote and won’t be upset if out voted. And Im happy to implement whatever the group decides.

hoechenberger, 2020-05-19:
Also makes you wonder how reasonable a 79-characters-per-line limit is these days, esp. considering a language like Python where indentation of blocks is enforced.

rob-luke, 2020-05-19:
>  If your research takes place in other domains, you can ignore the acronym, as you will not apply that method anyway…

This is the only thing I disagree with. I think promoting cross domain usage of methods is very important. Sharing knowledge and methods across domain is something we should encourage. We shouldn’t silo different domains, that’s the whole reason I want to use MNE, so I can apply eeg/meg processing to nirs with minimum effort.

hoechenberger, 2020-05-19:
Yes I totally agree with you. But instead of "acronyms", we could also just say "jargon" or "domain-specific wording" or… The thing is, even when terms and concepts are _not_ abbreviated, you often have not the slightest clue about what the heck they mean until you start reading up on this stuff. Temporal Derivative Distribution Repair is, IMHO, a prime example. This term tells me close to _nothing_. If anything, it makes me want to back off because it's such a monstrosity that's going to haunt my dreams and eat children alive, or something. :) just my 2 ct

rob-luke, 2020-05-19:
I would suggest to sticking to the discussion about length of names. The author wrote a great algorithm and probably put lots of consideration in to that name.

hoechenberger, 2020-05-19:
Just to clarify, I didn't mean to imply the name was bad. I just wanted to make the point that it's commonly the case that if you encounter a technical name or concept outside of you own domain, it typically requires quite some effort to grasp it. Whether it's an abbreviation or not doesn't really matter.

rob-luke, 2020-05-19:
There is much more information in the name than the acronym. It tells me it acts on temporal information, that it repairs the signal based on distribution information. 

If I put myself in the role of supervisor and am reviewing a students code where they have used a new approach i don’t know (this happens lots that students find the latest thing that I didn’t know about). While skimming through a students code looking for why the output looks strange, if I see the words repair and temporal it will inform me about what was done. Tddr will not. 

I think votes have been cast. There is no correct decision here. Someone just needs to make a decision, I’m new to MNE and don’t know the system, so will leave this to more experienced developers. I’ve appreciated the discussion though.

hoechenberger, 2020-05-19:
@drammock 
> is `temporal_deriv_distrib_repair` an acceptable compromise?

I think I could live with that. :)

rob-luke, 2020-05-19:
Somehow I missed @larsoner https://github.com/mne-tools/mne-python/issues/7798#issuecomment-631094350 above. Seems like a nice solution.

cbrnr, 2020-05-20:
Alright, it seems like we agree that acronyms should be used sparingly. However, the name `temporal_derivative_distribution_repair` is an extreme case - function names shouldn't be that long. Importantly, the acronym TDDR doesn't appear in isolation - it's always `mne.preprocessing.nirs.tddr`, so users immediately know that this is something specific to NIRS processing. Again, a user either knows the algorithm (thus also the acronym TDDR) or not. In the latter case, he/she will read at least the docstring in any case before using the function for the first time, so the function name is really not that important. In this case, shortening the length of the function name trumps verbosity IMO.

Let's try to come to a conclusion. These are the options people have proposed:

1. Use `tddr`
2. Use `temporal_derivative_distribution_repair`
3. Use `temporal_deriv_distrib_repair`
4. Use `temp_deriv_distrib_repair`
5. Make an alias `tddr = temporal_derivative_distribution_repair`

My vote is 1, but I could certainly live with 5, and maybe even 4 or 3. I'm strictly against 2. Could everyone cast their votes using these five options?

cbrnr, 2020-05-20:
(Fun fact: we're developing/using a package called MNE...)

hoechenberger, 2020-05-20:
> 1. Use `tddr` 👍👍
> 2. Use `temporal_derivative_distribution_repair` 👎
> 3. Use `temporal_deriv_distrib_repair`👍
> 4. Use `temp_deriv_distrib_repair`👎
> 5. Make an alias `tddr = temporal_derivative_distribution_repair`😐/ 👎

cbrnr, 2020-05-20:
@agramfort @larsoner @rob-luke @drammock could you also indicate your preferences?

agramfort, 2020-05-20:
if you can live with temporal_deriv_distrib_repair i would prefer but I will not fight a war for this. I need to choose my battles :)

drammock, 2020-05-20:
> 1. Use `tddr` 👎
> 2. Use `temporal_derivative_distribution_repair` 👎
> 3. Use `temporal_deriv_distrib_repair`👍
> 4. Use `temp_deriv_distrib_repair`👎
> 5. Make an alias `tddr = temporal_derivative_distribution_repair`👍

larsoner, 2020-05-20:
1. Use `tddr` 😐
2. Use `temporal_derivative_distribution_repair` 👍
3. Use `temporal_deriv_distrib_repair` 😐
4. Use `temp_deriv_distrib_repair` 👎
5. Make an alias `tddr = temporal_derivative_distribution_repair` 👍

cbrnr, 2020-05-20:
Summary:

1. 👎😐👍👍
2. 👎👎👎👍(👍)
3. 😐😐👍👍(👍)
4. 👎👎👎👎
5. 😐👍👍👍(👍)

The (👍) are inferred from comments: 2. @rob-luke, 3. @agramfort, 5. @rob-luke.

There's still no unanimously accepted solution, but 5 seems to be favored by most of us (or it seems to be the one with the fewest rejections at least). Using the mapping 👎=-1, 😐=0, and 👍=1, these are the total scores:

1. `tddr = temporal_derivative_distribution_repair`: 4
2. `temporal_deriv_distrib_repair`: 3
3. `tddr `: 1
4. `temporal_derivative_distribution_repair`: -1
5. `temp_deriv_distrib_repair`: -4

Personally, I'm happy with either the first or second place - can someone (@agramfort and/or @larsoner) please make the final decision between the alias and the slightly shortened function name?

hoechenberger, 2020-05-20:
(Just wanted to chime in and say I'm so happy we've got those Unicode icons! This makes fights about naming schemes so much more colorful! Loving it! 😍)

larsoner, 2020-05-20:
Let's do the alias, it allows the shortest-name and longest-name to be maximally happy

cbrnr, 2020-05-20:
OK! Where should I create the alias? In `__init__.py` or `_temporal_derivative_distribution_repair.py`? Or somewhere else?

larsoner, 2020-05-21:
I would put it in `_temporal_derivative_distribution_repair.py` then import it wherever we import the longer name. Then you'll probably also need to add it to the ignores in `test_docstring_parameters.py` (because it won't be in `python_reference.rst`), or we'll need to add it to `python_reference.rst` -- though I think I'd prefer just to have the long name documented there, then say in `Notes` of that function that a convenience alias is provided with `mne.preprocessing.tddr` or whatever.

PR #7800 (2020-05-22): Alias temporal_derivative_distribution_repair to tddr
Fixes #7798. I've also made the docstring of the internal `_TDDR` function more Pythonic.
closed 2020-05-22 by larsoner