mne-tools/mne-python #7799: Should viz.plot_raw_psd(..., tmin=None) behave differently?

hoechenberger, 2020-05-19:
Currently, if the user doesn't pass `tmin` to `viz.plot_raw_psd()`, it  is set to 0 if `n_fft is None`:
https://github.com/mne-tools/mne-python/blob/903f443e201f7dad94f0db456cac510cea6e31b4/mne/viz/raw.py#L626-L630

Now I'm wondering
- Why is it only set to 0 if `n_fft is None`
- Should `tmin` not be set to `raw.first_time` by default (i.e., when `tmin=None`) to yield more "relevant" results for most use cases?

larsoner, 2020-05-19:
> Should tmin not be set to raw.first_time by default 

No it shouldn't. The time the user passes should be relative to `raw.times[0]`, which is always zero.

As to why this is in the conditional and not outside it, I'm not sure, I'll have to look...

hoechenberger, 2020-05-19:
Thanks @larsoner. I also just had a phone call with @agramfort that helped clarify things a little on my end. :)

larsoner, 2020-05-19:
Okay, feel free to investigate the limits / conditionals further, let me know if you need help :)

closed 2020-07-20 by hoechenberger