mne-tools/mne-python #7805: ENH: Add ctrl+key or alt+key as another way to move in large steps.

seankmartin, 2020-05-21:
# Problem
There is a bug that can occur in the matplotlib tkinter backend that can prevent shift+key combo events from being correctly passed. A shift+key combo is used in mne interactive plots to move in large steps, and the tkinter backend bug can break this feature.

# Solution
A good enhancement would be to add the additional keybind ctrl+key or alt+key which performs the same functionality as shift+key in case a user has this bug.

# Alternative
Alternatively, using the pyqt5 backend for matplotlib also fixes this bug, but this would require pyqt5 to be installed.

# Additional context
This issue was discussed on gitter before posting here.

cbrnr, 2020-05-21:
Good idea!

larsoner, 2020-05-21:
+0.5 for ctrl+key over alt+key

rob-luke, 2020-05-22:
This is slightly off topic (so maybe I should open another issue?), but if someone is reorganising that section of the code it would be really handy to be able to use the modifier key to also make large changes in the time shown per view. e.g. if currently pressing `end` increases the time shown from 10 to 11 seconds, presssing `ctrl+end` would double the time view from 10 to 20 seconds.

cbrnr, 2020-05-22:
Also +0.5 for ctrl+key. I think this change and @rob-luke's suggestions can be tackled with one PR.