mne-tools/mne-python #7808: Request for a dedicated reader for Ricoh MEG data (.con files)

apesquita, 2020-05-22:
Hello,

I would like to suggest to make it possible to directly read Ricoh MEG system data into MNE python.
Please see the example file attached.

Many thanks!

Ana Pesquita
[010409_Motor_task_coregist-export_tiny.zip](https://github.com/mne-tools/mne-python/files/4666616/010409_Motor_task_coregist-export_tiny.zip)

agramfort, 2020-05-22:
do you know any open source reader for these files or someone who knows
well the format?

>

cbrnr, 2020-05-22:
It seems like Fieldtrip (http://www.fieldtriptoolbox.org/getting_started/ricoh/) and Brainstorm (https://neuroimage.usc.edu/brainstorm/) can read Ricoh data.

apesquita, 2020-05-22:
Hello,

Fieldtrip has some information on these files:
http://www.fieldtriptoolbox.org/getting_started/ricoh/
Is this helpful?

Thanks!

Ana P

agramfort, 2020-05-22:
we will need a BSD compatible implementation (no GPL code)

>

apesquita, 2020-05-22:
Hello,

I will try to get info about a BSD compatible implementation reader from the Ricoh company (and link you up with them).
In addition, here is the marker file that belongs to the .con data file.

Many thanks!

Ana P

 
[010409_marker.zip](https://github.com/mne-tools/mne-python/files/4667506/010409_marker.zip)

teonbrooks, 2020-05-23:
Ricoh is the new owners of the KIT/Yokogawa system. Unless their format has changed, one should be able to use the KIT module we have.

@apesquita, I was able to read in the file linked above. Try using the reader `mne.io.read_raw_kit`, let us know if this doesn't work for you.

agramfort, 2020-05-27:
indeed I tried:

`raw = mne.io.read_raw_kit('010409_Motor_task_coregist-export_tiny.con')`

and it works fine !

apesquita, 2020-05-27:
Thanks! I will check it asap with Kyung-Min (somehow I missed this message before). We will keep you posted. Many thanks!

apesquita, 2020-06-07:
Hello all,

Thank you for your suggestions.
Indeed the mne.io.read_raw_kit seems to open the file.

But when using the mne.viz.plot_alignment we get back a geometry of sensors and helmet that seem to be somewhat misaligned. Please see the pdf in the attachment. 

Have you encountered this issue before?

Many thanks!

[image]

agramfort, 2020-06-07:
one cannot see attachments on github

>

apesquita, 2020-06-07:
> one cannot see attachments on github
> […](#)

Updated previous message. Thanks!

Kyung-minAn, 2020-06-08:
@agramfort & @apesquita 
This is Kyung-min.
I'm glad to hear that you would make an MNE tutorial for Ricoh data. 
The data is acquired using a child-customized MEG system located in Kanazawa, Japan.
I attach the raw MEG data with auditory stimuli and Freesurfered MRI data by email.

Thanks a lot!!

agramfort, 2020-06-08:
@Kyung-minAn you cannot attach files on github.

can you make the files available on dropbox or google drive or whatever works?

agramfort, 2020-06-09:
I started to look into. I can produce meaningful evoked data but I have some issues:
- the KIT reader overwrites the native channel names so using the shared layout files is hard. @teonbrooks or @christianbrodbeck do you know if it's possible to avoid having this forced renaming to MEG001, MEG002 etc...
- should we introduce a new read_raw_xxx function. If so how would you name it? raw_raw_yokogawa, raw_raw_ricoh, raw_raw_con?

anyone would have time to look into this? It's getting too ambitious for me to looking into it myself given the time I have.

apesquita, 2020-06-09:
Hello,

Many thanks for these developments.
I was wondering if it would be the moment to get the Ricoh engineers in touch with the mne developers?
Min had forwarded some questions from Ricoh to the mne developers:

  1) MNE-Python has 'mne.io.*' functions depending on the type of MEG system to read the data.
   Are you planning to make a 'mne.io.ricoh.*' function?

2) If so, could you please tell me what Ricoh company contributes to it?
   They may provide a primitive script to read binary data ('*-export.con file'), although they cannot contribute much to the development of io/ricoh itself since they are not so familiar with fiff format.

Many thanks!

Ana P

agramfort, 2020-06-09:
if we can have a Ricoh engineer to help it would be great !

>

Kyung-minAn, 2020-06-11:
@agramfort 
I sent e-mail with a contact of Ricoh engineer.
Please check it !
And thanks a lot~!!

teonbrooks, 2020-06-11:
If we can get a clear idea of what would be needed, I might be able to dedicate some spare cycles to get this work and/or coordinate with Ricoh engineers to modify our existing module. 

I wonder what the best nomenclature for the io package should be for Ricoh/Yokogawa/KIT. I 

>     * the KIT reader overwrites the native channel names so using the shared layout files is hard. @teonbrooks or @christianbrodbeck do you know if it's possible to avoid having this forced renaming to MEG001, MEG002 etc...

for the renaming, do we accept MEG channels that don't begin with `MEG`? I know that Fieldtrip naming for the channels are prefixed with AG for the axial gradiometers Ricoh hardware uses.

As for the layouts, can we derive the layout for the MEG system like we do for the EEG ones now?

larsoner, 2020-06-11:
> for the renaming, do we accept MEG channels that don't begin with MEG

Some `planar1`/`planar2` can break if the naming is wrong for Neuromag systems, but I don't know of any restrictions for other systems. It's best if we can always use the manufacturer's naming as much as possible. So indeed we should not rename things as `MEG*` if they are not named this way on the acquisition system.

agramfort, 2020-06-11:
I have layout files for Ricoh systems like for other MEG systems. We
'just' need the right channel names :)

larsoner, 2020-06-11:
@teonbrooks want to take a stab at reading the correct names? Maybe add a `rename=None` param that defaults to `True` in 0.21 (do what we do now) but will be `False` in 0.22 (use the native names) via deprecation?

teonbrooks, 2020-06-11:
sounds good. yeah, I take a stab at this over the weekend

teonbrooks, 2020-06-17:
I took a look at this, and we're generating the name of the channels by their index. We don't have the original label information anywhere. It would be super helpful if we could be in touch with someone at Ricoh to see what block in the data may have these labels so we could read them directly. @agramfort, has the Ricoh engineer reached to you yet? If you would like, I could coordinate with them and see if they have this info anywhere. For Ricoh system in FieldTrip, they're using pre-compiled private matlab scripts. It's likely they could share with us how they do so if they have the source files.

agramfort, 2020-06-17:
let me put you in contact with ricoh engineer

>

apesquita, 2020-06-18:
Thank you for developing this reader! I will be thrilled to use it later on!

apesquita, 2020-07-31:
Dear Alexander and Teon,

Did you receive all the necessary information from the Ricoh engineers to put together the mne reader for this system?
Do you think that it will be feasible to develop the mne reader for Ricoh?
Many thanks for any updates on this issue.

All the best!

Ana P

agramfort, 2020-07-31:
hi,

I started to look and add the reading of the channel names but it will need
more time

I asked for help to @larsoner

we'll make progress in the coming weeks. I think we have all the info we
need and
I could run the code shared with us.

thanks for your patience

>

Kyung-minAn, 2020-08-02:
Thank you for developing mne readers.
I'm excited about the great news!!
I appreciate your contributions~

Looking forward to hearing from you~!!

Best wishes,
Min

larsoner, 2020-08-05:
@apesquita channel name reading seems to be working and we can just generate a layout on the fly for your data (with the right constants). I'm at the point that it would help to have a subject's evoked dataset plus MRI to do some source localization on to test alignment, etc.

larsoner, 2020-08-24:
This should be fixed in `master`

closed 2020-08-24 by larsoner

Kyung-minAn, 2021-07-22:
Hello. @agramfort @larsoner 
I hope you are doing well.

We (with @apesquita ) finally met in Birmingham and tried to import our data from the child-customized MEG at Kanazawa, Japan.
We can read the raw con data using 'mne.io.read_raw_kit', but not the coregistrated data.

The error code is below:

`import mne
raw = mne.io.read_raw_kit('010409_Motor_task_coregist-export_tiny_1s.con')
Extracting SQD Parameters from 010409_Motor_task_coregist-export_tiny_1s.con...
Creating Raw.info structure...
Traceback (most recent call last):

  File "C:\Users\eiren\AppData\Local\Temp/ipykernel_9800/996202582.py", line 2, in <module>
    raw = mne.io.read_raw_kit('010409_Motor_task_coregist-export_tiny_1s.con')

  File "C:\Users\eiren\anaconda3\envs\mne\lib\site-packages\mne\io\kit\kit.py", line 908, in read_raw_kit
    return RawKIT(input_fname=input_fname, mrk=mrk, elp=elp, hsp=hsp,

  File "<decorator-gen-234>", line 24, in __init__

  File "C:\Users\eiren\anaconda3\envs\mne\lib\site-packages\mne\io\kit\kit.py", line 133, in __init__
    info, kit_info = get_kit_info(

  File "<decorator-gen-237>", line 24, in get_kit_info

  File "C:\Users\eiren\anaconda3\envs\mne\lib\site-packages\mne\io\kit\kit.py", line 749, in get_kit_info
    assert meg_done

AssertionError`

I am looking forward to hearing from you.

Many thanks,
Min

larsoner, 2021-07-22:
Can you share the data with us (link here or over email)?

Kyung-minAn, 2021-07-22:
@larsoner Thank you for your prompt response!! 
I share a short version of our coregistrated data here. 

[010409_Motor_task_coregist-export_tiny_1s.zip](https://github.com/mne-tools/mne-python/files/6864118/010409_Motor_task_coregist-export_tiny_1s.zip)

Thank you for all your effort!