mne-tools/mne-python #7813: Raw.anonymize() docstring is irritating

hoechenberger, 2020-05-23:
When viewing the docs of `Raw.anonymize()`, I'm confronted with a warning:

> If info is part of an object like raw.info, you should directly use the method raw.anonymize() to ensure that all parts of the data are anonymized and stay synchronized (e.g., raw.annotations).

which is irritating because, well, I **am** reading up on `Raw.anonymize` already. Also see screenshot below.

[image]

hoechenberger, 2020-05-23:
Also… what's a "date of service"?

agramfort, 2020-05-23:
date of service can refer to the date at which the MEG system was checked
by MEGIN.
It's typically once a year.

>

larsoner, 2020-05-24:
Feel free to modify the docstrings using the `docdict` so that the warning only shows up in the right place

bloyl, 2020-05-24:
Date of service, in this context refers to `info['meas_date']`

agramfort, 2020-05-24:
I would consider a rephrasing then. Date of acquisition ?

closed 2020-07-20 by hoechenberger