mne-tools/mne-python #7814: Azure Windows 3.7 conda job keeps failing

cbrnr, 2020-05-23:
For quite some time the Azure Windows 3.7 conda job has been throwing the following error:

```
FAILED mne/minimum_norm/tests/test_inverse.py::test_make_inverse_operator_diag[testing_data-testing_data]
```

More detailed log message:

```
================================== FAILURES ===================================
_________ test_make_inverse_operator_diag[testing_data-testing_data] __________
[gw0] win32 -- Python 3.7.4 D:\a\1\s\conda\python.exe
mne\minimum_norm\tests\test_inverse.py:727: in test_make_inverse_operator_diag
    rtol=1e-1, atol=1e-1, ctol=0.99, check_K=False)
mne\minimum_norm\tests\test_inverse.py:182: in _compare_inverses_approx
    assert corr > ctol
E   assert 0.7766994175372354 > 0.99
```

@larsoner suspects that this is related to MKL and probably hard to track down.

agramfort, 2020-05-25:
@larsoner any thought?

>

larsoner, 2020-05-25:
It seems to be somewhat sporadic, I've only seen it a handful of times. Not sure how that's possible, unless somewhere memory is getting corrupted. I can try replicating but want to see if it shows up again, since the last couple of PRs I looked at did not show a problem...

cbrnr, 2020-05-27:
https://dev.azure.com/mne-tools/mne-python/_build/results?buildId=7369&view=logs&j=a846d25a-e32c-5640-1b53-e815fab94407&t=f8799db5-b972-5f58-516d-bc88b8357f99

larsoner, 2020-05-27:
So it is non-deterministic. Great -- to my windows boot I go, wish me luck...

GuillaumeFavelier, 2020-06-29:
What is the current situation on this @cbrnr ? Probably many things have changed since the issue has been opened.

cbrnr, 2020-06-29:
I haven't seen this lately (but I didn't have too many PRs either). @larsoner?

larsoner, 2020-06-29:
I haven't seen it in a while, let's optimistically close and reopen if need be

larsoner, 2020-07-01:
Not dead, happened in #7952

larsoner, 2020-08-24:
Okay maybe it's fixed now...

closed 2020-08-24 by larsoner