mne-tools/mne-python #7818: Use properties to set values in Info

cbrnr, 2020-05-24:
As I've mentioned in #7816 I think it would be more intuitive if users could just set values in the `Info` dict using simple assignments, e.g. `info['meas_date'] = None`, instead of setter methods (e.g. `info.set_meas_date(None)`). This is consistent with values that we currently allow to be set directly, e.g. `info["bads"] = []`.

This would also allow us to make read-only properties for fields which users should not set manually.

agramfort, 2020-05-24:
my personal feeling: If everytime someone does something bad you create a
law you end
up with 10 new laws every month and you end up with a non-agile system as
you have
a lot to consider when amending the law. It's the same thing with
software/code. Trying
to prevent you every possible mistake will make you design a monster and
will slow down
all the rest (more code to maintain, more things to test etc.)

>

cbrnr, 2020-05-24:
I totally agree with you - that's why I'd use properties instead of setters.

hoechenberger, 2020-05-25:
I agree with @cbrnr: Using properties would allow us to ensure "correct" usage –  _without_ having to invent and apply arbitrary rules. I think it makes sense to have this for the `info` structure, as it's one of the bits that users actually are expected to touch and modify. I don't think anyone is wanting to expand this to other parts of MNE for now. :)

cbrnr, 2020-05-25:
No, this is just about the `Info` object. Intended changed usage examples:

```python
>>> raw.info["bads"] = ["C3", "Cz", "C4"]
>>> raw.info["annotations"] = mne.Annotations(mne.Annotations(5, 1, "test"))
>>> raw.info["sfreq"] = 254
AttributeError: sampling frequency cannot be changed (consider resampling the signal)
>>> raw.info["ch_names"] = ["X", "Y"]
ValueError: number of channel names (2 given) must match number of channels (32)
>>> raw.info["nchan"] = 30
AttributeError: number of channels cannot be set (consider using drop_channels)
```

agramfort, 2020-05-25:
I propose to sleep over this for a few weeks and concentrate on more
blocking matters :)

>

cbrnr, 2020-05-25:
Hehe, in case you're hoping that I will forget: I won't 😄.

But I won't insist, if you say we don't need it - fine with me. I'd be interested what others think - especially from a user's point of view. Any opinions?

hoechenberger, 2020-05-26:
> I'd be interested what others think - especially from a user's point of view. Any opinions?

I was going to tag some users here, but I suppose not all of them would even know what this "properties" thing is supposed to be and when and how it would affect their work. If you feel like it, could you maybe write a short example to showcase what it is you mean? Then we could get users' opinions :) (You already have sth in your initial comment here, but its phrasing is IMHO mainly targeted towards developers still!)

cbrnr, 2020-05-26:
Actually I just realized that the only setter method for `Info` objects is `set_montage`. So maybe documenting which fields are user-writable is sufficient.

agramfort, 2020-05-26:
>
> Actually I just realized that the only setter method for Info objects is
> set_montage. So maybe documenting which fields are user-writable is
> sufficient.
>
+10

hoechenberger, 2020-05-26:
> So maybe documenting which fields are user-writable is sufficient.

And still using properties to avoid touching items you're not supposed to touch? Or what do you mean? :)

cbrnr, 2020-05-26:
> And still using properties to avoid touching items you're not supposed to touch? Or what do you mean? :)

Nope, just document which fields are supposed to be touched and that's it.

hoechenberger, 2020-05-26:
Ok, sounds good! Just try to make sure it's, like… really, **really** in-your-face 😁

cbrnr, 2020-05-26:
Yes, the note should be promoted to a warning or something.

PR #7832 (2020-05-26): Clearer info docstring
Fixes #7818. @hoechenberger @larsoner @agramfort OK with you?
closed 2020-05-26 by larsoner