mne-tools/mne-python #7829: scalings on CSD eeg signals doesn't work

ZQInQAAS, 2020-05-26:
when I try this tutorial[https://mne.tools/dev/auto_examples/preprocessing/plot_eeg_csd.html?highlight=laplacian](url) I found parm **scalings** doesn't work on raw_csd's eeg. Here's my code:
`raw_csd = mne.preprocessing.compute_current_source_density(raw)`
`raw.plot(scalings=dict(eeg=20, eog=500))`
`raw_csd.plot(scalings=dict(eeg=3000, eog=200))`
[image]
left is raw, right is raw_csd.  
But **scalings** on raw_csd's eog works well. So I think it's maybe a bug? After CSD the data cannot be picked by 'eeg' label?  Looking forward to your reply.  0.<

larsoner, 2020-05-26:
Perhaps you need to pass `scalings=dict(csd=...)` ? Not sure...

ZQInQAAS, 2020-05-26:
wow……It's really works!  Thank you~

closed 2020-05-26 by ZQInQAAS