mne-tools/mne-python #7834: FIFF file exceeded 2GB limit, please split file or save to a different format

SophieHerbst, 2020-05-27:
I am getting this error for an epochs.fif file when trying to save it:
> FIFF file exceeded 2GB limit, please split file or save to a different format

The same files for other subjects are automatically split and saved (2 x 1.2 GB), but for this one I have a 2.1 GB file on my disk and then get the error.

Is there a way to force the splitting? Thanks!

agramfort, 2020-05-27:
a bug about this was recently reported. Can you try with latest stable
release?
is it epochs or raw?

>

SophieHerbst, 2020-05-27:
it is epochs and I am using mne 0.21.dev0

agramfort, 2020-05-27:
update to 0.20.5 or current master and it will fix your pb (I think)

>

SophieHerbst, 2020-05-27:
fixed by update to current master

closed 2020-05-27 by larsoner