mne-tools/mne-python #7836: Possible bug in _pick_data_or_ica

cbrnr, 2020-05-27:
While setting `meg=True` in calls to `mne.pick_types` in calls that didn't explicitly specify this argument (#7823), I noticed that the internal function `_pick_data_or_ica` probably doesn't do the right thing.

Previously, the function picked:

```python
picks = pick_types(info, exclude=exclude, misc=True)
```

Since `meg=True` by default, I changed this to:

```python
picks = pick_types(info, meg=True, exclude=exclude, misc=True)
```

But this function is supposed to pick ICA channels (I think), and these should be of type `'misc'` (another assumption I'm not sure of). But if this is true, `meg` should be set to `False`.

larsoner, 2020-08-24:
This looks like it should be `meg=False`. @cbrnr can you try making this change and see if it breaks any existing test?

cbrnr, 2020-08-24:
Will do!

PR #8155 (2020-08-25): Remove meg=True from _pick_data_or_ica
Fixes #7836.
closed 2020-08-25 by agramfort