mne-tools/mne-python #7837: Magnetometer positions look odd in PSD plot

hoechenberger, 2020-05-27:
#7163 ## Describe the bug
〈Replace this text with a clear and concise description of the bug.〉

#### Steps to reproduce
Let's first visualize the magnetometer channel positions via `Raw.plot_sensors()`:
```python
import mne
from mne.datasets import sample

data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
raw.plot_sensors(ch_type='mag')
```
This produces:
[image]

Looking good. Now let's plot PSDs of the data.

```python
raw.plot_psd()
```
Result cropped so it only shows the sensor positions (top to bottom: mag, grad, eeg):
[image]

The magnetometer sensor layout looks incorrect, and doesn't match the one produced by `plot_sensors()`. Am I missing something here?

agramfort, 2020-05-27:
to me yes we have a bug.

any idea @larsoner ?

>

larsoner, 2020-05-27:
Agreed it's a bug, no idea the cause. Someone will have to dig into the viz code to figure out why this is happening...

agramfort, 2020-06-08:
this allows to clearly show the bug:

```
import mne
import os

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, preload=True)
raw.plot_psd(picks=['grad'])
raw.plot_psd(picks=['mag'])
raw.plot_psd(picks=['meg'])
```

agramfort, 2020-06-08:
and the bad news it that the bug is in the 0.20 stable release :( :(

larsoner, 2020-06-08:
I'll `git bisect` and see how hard it is to fix

larsoner, 2020-06-08:
Looks like it was 9e785061f325191b809ade41590887230c1378ec / #7486, cc @drammock -- I'll see if I can figure out what the problem is

PR #7878 (2020-06-08): MRG, FIX: Fix get_channel_types
Fixes a bug with `plot_psd` that was introduced by #7486. Basically the order in `picks` was not preserved, which was required by `plot_raw_psd` when there are multiple channel types.

Closes #7837
closed 2020-06-08 by larsoner