mne-tools/mne-python #7838: Numbers appended to channel names from CTF data.

gbrookshire, 2020-05-27:
#### Describe the bug
Reading data with `mne.io.read_raw_ctf` appends numbers to the end of the channel names.

#### Steps to reproduce
I'm using this example data from Fieldtrip: ftp://ftp.fieldtriptoolbox.org/pub/fieldtrip/tutorial/SubjectCMC.zip

```python
import mne
raw = mne.io.read_raw_ctf('../data/SubjectCMC/SubjectCMC.ds')
raw.ch_names[-10:]
```

#### Expected results
I expect to see the same channel labels that show up when reading the data into Fieldtrip. Here are the last few channel names:
```
    {'MZO01' }
    {'MZO02' }
    {'MZP01' }
    {'MZP02' }
    {'EOG'   }
    {'ECG'   }
    {'ADC15' }
    {'ADC16' }
    {'EMGlft'}
    {'EMGrgt'}
```

#### Actual results
```
['MZO01-1706',
 'MZO02-1706',
 'MZP01-1706',
 'MZP02-1706',
 'EMGlft',
 'EMGrgt',
 'EOG',
 'ECG',
 'ADC15-406',
 'ADC16-406']
```

(It's also worth noting that they're not in the same order as in Fieldtrip, but I don't have any idea whether that's a bug.)

#### Additional information
```
mne.sys_info()                                                                                                
Platform:      Linux-4.15.0-99-generic-x86_64-with-debian-buster-sid
Python:        3.7.7 (default, Mar 26 2020, 15:48:22)  [GCC 7.3.0]
Executable:    .../anaconda3/envs/mnedev/bin/python
CPU:           x86_64: 4 cores
Memory:        39.1 GB

mne:           0.21.dev0
numpy:         1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:         1.4.1
matplotlib:    3.1.3 {backend=Qt5Agg}

sklearn:       0.22.1
numba:         0.49.1
nibabel:       3.1.0
cupy:          Not found
pandas:        1.0.3
dipy:          1.1.1
mayavi:        4.7.1
pyvista:       0.24.2
vtk:           8.2.0
PyQt5:         5.9.2
```

larsoner, 2020-05-27:
I don't know enough about CTF to know. @bloyl any insight here?

agramfort, 2020-05-27:
yes the ch_name you have in the file. We prefer not to drop them but we
should be
able to ignore the numbers when needed (plotting etc.)
We have a function called _clean_names used internally for this

>

larsoner, 2023-10-01:
Nowadays we have `rename_channels` for this, but hopefully the fuller/hopefully-more-correct values are usable in most use cases without renaming

closed 2023-10-01 by larsoner