mne-tools/mne-python #7850: viz.plot_sensors() only partially honors axes parameter

hoechenberger, 2020-05-29:
#### Describe the bug
I wanted to add sensor alignment plots via `Raw.plot_sensors()` to some subplots of a larger figure. While the sensors **are** plotted, I observe some very annoying side effects that imho qualify as bugs.

#### Steps to reproduce
```
import matplotlib.pyplot as plt

import mne
from mne.datasets import sample

data_path = sample.data_path()
fname = data_path + '/MEG/sample/sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)

fig, ax = plt.subplots(2, 2)

ax[0, 0].plot(range(10))
ax[0, 1].set_title('Title A')

ax[1, 0].plot(range(10))
ax[1, 0].set_title('Title B')

raw.plot_sensors(ch_type='mag', axes=ax[0, 1])
raw.plot_sensors(ch_type='grad', axes=ax[1, 1])
```

#### Expected results
All subplots should appear in their designated axes.

#### Actual results
[image]

- First sensor plot doesn't have a title
- Second sensor plot's title appears as `suptitle` while it should be an axes title
- The entire figure appears to be truncated at the top, left, and bottm

#### Additional information
The behavior reminds me somewhat of #7595, which has been fixed by @drammock in #7706.

hoechenberger, 2020-05-29:
There also seems to be an issue with the "click on sensors" callback, which is linked across BOTH sensor subplots…

[image]

agramfort, 2020-05-29:
I would qualify this as a bug

>

PR #8536 (2020-11-18): MRG, VIZ, FIX: plot_sensors title and interactivity
closes #7850 

- `plot_sensors` now puts title as axes title instead of `fig.suptitle`
- the on_pick callback now won't act unless the click was in the correct axes
closed 2020-11-18 by larsoner