mne-tools/mne-python #7854: A potential mismatch of times rescale in subtract_evoked

listenzcc, 2020-06-01:
**READ THIS FIRST:** If you are having trouble getting MNE-Python to work with
your own data, you should probably ask for help on our
[Gitter channel](https://gitter.im/mne-tools/mne-python) or on our
[mailing list](https://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis)
before opening an issue; you will get help faster that way. See
[the FAQ entry on filing bug reports](https://mne.tools/dev/overview/faq.html#i-think-i-found-a-bug-what-do-i-do)
for more guidance. If you're pretty sure your problem is a bug in MNE-Python,
**delete this section** and fill in the headings below, replacing the
placeholder text with your own responses. Surround any code samples with triple
backticks above and below the code block, or put code in a
[public gist](https://gist.github.com) and link to it in the issue.

#### Describe the bug
In epochs analysis, a common operation is to compute induced response by subtracting the mean epochs. In MNE-python, it is computed by _detrend_offset_decim function in epochs sub-module.

However, the time scale operation seems use self._raw_times as time line instead of self.times.

The two values may different if an user used DECIM parameter when they get epochs, the self._raw_times is keep unchanged and self.times has been recomputed according to DECIM parameter. If the mismatch happens, the program will crush immediately, and raise a ValueError says operands could be broadcast together with shapes x, y, x. (x means original times in self._raw_times, and y means new times in self.times)

#### Steps to reproduce
〈Replace this text with a code snippet or minimal working example [MWE] to
replicate your problem, using one of the [built-in datasets], preferably the
one called [sample]. If you can't replicate on a built-in dataset, provide also
a link to a small, anonymized portion of your data that does yield the error.〉

# Get a raw.fif from somewhere
raw = mne.io.read_raw_fif('[some raw.fif file]')
# Get Epochs using DECIM parameter
DECIM = [some value other than 1]
epochs = mne.Epochs(raw, events=[events], decim=DECIM)
# Compute induced response
# ****** It will crush **********
epochs_induced = epochs.copy().subtract_evoked(evoked)

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
〈Replace this text with a clear and concise description of what you expected to
happen.〉

#### Actual results
〈Replace this text with the actual output, traceback, screenshot, or other
description of the results.〉

#### Additional information
〈Replace this text with the output of `mne.sys_info()`.〉

drammock, 2020-06-01:
Please read carefully the instructions in the bug report template, paying special attention to the parts that say "delete this section" or "replace this text with...". In particular, pay attention to the request that your code sample be *reproducible* so that others can copy/paste your code and have it work *without needing to edit it* (i.e., avoid placeholders like "some raw.fif file").

agramfort, 2020-06-01:
I confirm the bug. You can go around it for now by using preload=True in Epochs

PR #7855 (2020-06-01): MRG, FIX: Fix subtract_evoked with decim
Closes #7854.

@listenzcc can you see if it fixes your problem?
closed 2020-06-01 by larsoner