mne-tools/mne-python #7857: EEG Emotiv and other .edf files:  ValueError: DigMontage is only a subset of info. There are 45 channel positions not present in the DigMontage. The required channels are:

barlehmann, 2020-06-01:
I have been wrangling with problems around digitization points and sensor locations for EEG files for weeks. I have read through the sensor locations tutorial several times on the MNE website but have not yet been able to fix the issue. I have looked at a number of github issue pages with related issues but have not found any which have helped to resolve this issue. I am using .edf files from Emotiv Flex (32channel) as well as other 19-32 channel devices and their respective .edf files. I am able to plot the raw data without issue, but digitization points come out as "None". When I set montage using: montage_kind = "standard_1005"
montage = mne.channels.make_standard_montage(montage_kind)
raw.set_montage(montage, match_case=False)
I get: ValueError: DigMontage is only a subset of info. There are 45 channel positions not present in the DigMontage. The required channels are: ['TIME_STAMP_s', 'TIME_STAMP_ms', '....'CQ_Cz', 'CQ_Fz', 'CQ_Fp1', '....]
I have also tried the same with another file with the required channels being more typical channel names, but I continue to get this same error whenever I try to set the montage, plot topomap, or other functions of basic EEG analyses. It is likely that the .edf files I am working with do not have montage or digitization points associated with them, but the problem is that I cannot seem to set the montage either - whenever I try I get this above "ValueError: DigMontage is only a subset of info" (despite the fact that the channels are clearly visible in the channel names in the raw.info). Any suggestions would be greatly appreciated! (Please keep in mind I am a beginner in MNE, so simplicity will also be greatly appreciated.)

larsoner, 2020-06-01:
On 0.20.5 you can do raise_if_subset=False and on 0.21 (master) you can do on_missing='ignore' to get past this error.

But it sounds like something more fundamental is wrong here. You need all EEG channels to have positions for anything topomap-related to plot properly, so using these on_missing is in general discouraged.

Channel types are an important concept in MNE, and it seems like you have channels that are not typed properly. For example, 'TIME_STAMP_s' does not seem like an EEG channel, so it seems like it should be set to 'misc' type using raw.set_channel_types. Once you set all non-EEG channels to be non-EEG type (like 'misc'), and you have a montage with all EEG channel names present, you should be able to do raw.set_montage(complete_montage, match_case=False) and things should work again.

@barlehmann where could we add to / where did you look in the tutorials where we could make this clearer? Maybe very early like:

https://mne.tools/dev/auto_tutorials/intro/plot_10_overview.html#loading-data

Setting channel types improperly can break / produce unexpected results in several steps, including Preprocessing (the next after "Loading Data"), so it seems like we should at least have a link there. @drammock WDYT?

drammock, 2020-06-01:
Quoting from https://github.com/mne-tools/mne-python/issues/4852#issuecomment-636982793

> Setting channel types improperly can break / produce unexpected results in several steps, including Preprocessing (the next after "Loading Data"), so it seems like we should at least have a link there. @drammock WDYT?

This is the trade-off between having clean sample data (common problems like this don't arise) and having messy sample data (you have to clean it every time you use it in a tutorial).  I see three natural-ish places to add this info:

1. add a separate tutorial to the intro section, called "working with different channel types", and in (only) that tutorial, load in a few different datasets with different issues to illustrate how to handle them.
2. "working with sensor locations" could maybe change to "channel types and sensor locations".
3. incorporate it into each of the tutorials in the section on "reading data for different recording systems"

barlehmann, 2020-06-02:
larsoner, thank you for your swift and useful suggestions - that is all helpful though I am still in need of clarification. My MNE version is .21 and I first tried using on_missing='ignore' and this allowed me to get a result for digitization points, however I am still unable to proceed. I totally agree with your point about the misnamed channels, but this does not seem to be at the root of the issue I seem to be experiencing. To show that this strange naming I had in the above message is not the issue, I switched to another EEG raw .edf file whose names only included those that seem to be on the 1020 montage: ['EEG Fp1-LE', 'EEG Fp2-LE', 'EEG F3-LE', ....]. However when I do so I receive a notification that:
 (1)
DigMontage is only a subset of info. There are 20 channel positions not present in the DigMontage. The required channels are: ['EEG Fp1-LE', 'EEG Fp2-LE', 'EEG F3-LE', 'EEG F4-LE', 'EEG C3-LE', 'EEG C4-LE', 'EEG P3-LE', 'EEG P4-LE', 'EEG O1-LE', 'EEG O2-LE', 'EEG F7-LE', 'EEG F8-LE', 'EEG T3-LE', 'EEG T4-LE', 'EEG T5-LE', 'EEG T6-LE', 'EEG Fz-LE', 'EEG Cz-LE', 'EEG Pz-LE', 'EEG A2-A1']
(2)
KeyError: 'eeg fp1-le'

These notifications and error messages are confusing to me since if I type in print(raw.info['ch_names']), I get the same exact list of channels that are specified as required - suggesting there is no discrpancy between the expected channel names and those that are in the raw.info. Furthermore, I am using match_case=False while setting the montage so I am not sure why this would also not be preventing the KeyError:  'eeg fp1-le' which seems to suggest that the expected lowercase/case setting of this sensor has not been met. 

Thank you again for your response and I look forward to hearing your thoughts,
Bar

agramfort, 2020-06-02:
default montage requires your channels to be called 'Fp1' and not 'EEG
Fp1-LE'

hope this helps

barlehmann, 2020-06-05:
Alexandre - thank you so much - that was very helpful.

Also Mr. Larson, if you would like to work on clarifying this - I had been
familiar with the word 'montage' referring to the arrangement that
electrodes are placed on the head and then referenced -  such as
'laplacian' or 'average-reference'  - in this case of MNE, montage and
digitization points were referring to spatial coordinates. Perhaps I am a
newbie and so others would not have been confused by this, but this was
certainly quite a challenging thing to understand firstly the definition
and necessity of the MNE Montage, and secondly the problem with the '-le'
ending of 'fp1-le' that was the automatic name I had in my .edf file.

larsoner, 2020-08-31:
We have `raise_if_subset` in 0.20 and `on_missing` in 0.21

https://mne.tools/dev/generated/mne.io.Raw.html#mne.io.Raw.set_montage

Closing but let's open if there is a remaining problem

closed 2020-08-31 by larsoner

PrometheusAtlas, 2023-03-01:
Ya, its 2023 and this is still happening. Suggestion: cut the developer lingo and actually sort out the problem. We are not here to read a conversation as to where a tutorial about a non solution should be.

cbrnr, 2023-03-01:
Can you share a minimal example so that we can reproduce your issue?

PrometheusAtlas, 2023-03-02:
> 

Hi, thanks for response. The issue is resolved for me but I can tell you for future reference or others that may get here again: when the error occurs for e.g. the 1020 standard montage the value error message says that the channels are not called 'EEG Fp1-Ref'. However when you try to debug by looking at the channel names to see what they are called, they are correct. So the error message is in fact misleading. As mentioned above, what is actually required is that the channels be changed from 'EEG Fp1-Ref' to just 'Fp1'. I did this using the following two lines of code: 
raw.rename_channels(lambda x: x.replace('-Ref', ''))
raw.rename_channels(lambda x: x.replace('EEG ', '')). Following this, the remaining value error is for the two non EEG channels which I then used the standard "raw.set_montage(montages, on_missing='ignore')" and all is working now thankfully. I imagine I will just remove those two channels in the end but I am just ignoring for now and I can create my bipolar montage in peace. Godspeed.

cbrnr, 2023-03-02:
Thanks. Please note that if you have an EDF file with "EEG" prefixed to channel labels, you can set `infer_types=True` in `mne.io.read_raw_edf()`. That way, the prefixes will be removed, and in your case, channel "EEG FP1-Ref" will be renamed to "FP1-Ref".

PrometheusAtlas, 2023-03-02:
Hi , no worries. Yes ok , but I don't think that would work as 'Fp1-Ref' is not 'Fp1'. I am open to correction on that as I have not tried it with 'Fp1-Ref' - I am removing bot EEG and Ref with my solution above.

cbrnr, 2023-03-02:
Yes, you will need to manually rename in this case.

PrometheusAtlas, 2023-03-02:
No, in all cases you will have to rename as there are the prefix 'EEG' and post fix in all datasets. So "infer_types=True" is useless for this issue and does not resolve it. If 'infer_types=True' was the solution it would have been provided already.