mne-tools/mne-python #7858: MNE Fiducials and Their Usage in mne.coreg.get_mni_fiducials

adam2392, 2020-06-01:
#### Problem
II'm trying to use the `mne.coreg.get_mni_fiducials` function, but it is not clear what the output is. Is the dictionary ordered as LPA, nasion, RPA? If not what is it?

In addition, I was wondering what the best way of attaching these coordinates to a raw is? Would one save the fiducials say as JSON and can then reload and attach during the set_montage step?

#### Suggested glossary term
It would be great to see a glossary item explaining how fiducials are used in general (or links) and how it is used in mne-python.

agramfort, 2020-06-01:
fiducials are necessary to define the head coordinate system. What this
function does it transform to subject head the
fiducials hard coded for fsaverage

clear enough?

>

vagechirkov, 2021-03-19:
@agramfort @adam2392 I think there is already [description in the glossary](https://github.com/vagechirkov/mne-python/blame/main/doc/glossary.rst#L123) and proper [documentation of the function](https://github.com/vagechirkov/mne-python/blob/56e9fbe67a554272a71dfd2b9ae96eb7fd58d104/mne/coreg.py#L1224). Is it still an open issue?

agramfort, 2021-03-19:
@adam2392 <https://github.com/adam2392> are the links shared above clear
enough?

adam2392, 2021-03-19:
Yes I think this is stale now. Can close

closed 2021-03-19 by adam2392