mne-tools/mne-python #7860: 0.20.4: Tests failing due to scikit-learn deprecation warnings

sanjayankur31, 2020-06-02:
#### Describe the bug
On Fedora 33, we have two tests failing due to deprecation warnings from scikit-learn.

mne-python-0.20.4. I'm looking at updating the Fedora package to 0.20.5 now.

#### Steps to reproduce
Run tests: ```pytest```

#### Expected results
Tests don't fail due to the deprecation warnings.

#### Actual results
Tests fail. Complete log attached.

- we're running scikit-learn-0.23.0 here on Fedora 33 and scikit-learn seems to force warnings: https://github.com/scikit-learn/scikit-learn/issues/2531

#### Additional information

Full failure log:

```
=================================== FAILURES ===================================
________________________________ test_get_coef _________________________________

    @requires_sklearn
    def test_get_coef():
        """Test getting linear coefficients (filters/patterns) from estimators."""
        from sklearn.base import TransformerMixin, BaseEstimator
        from sklearn.pipeline import make_pipeline
        from sklearn.preprocessing import StandardScaler
        from sklearn import svm
        from sklearn.linear_model import Ridge, LinearRegression
        from sklearn.model_selection import GridSearchCV

        lm_classification = LinearModel()
        assert (is_classifier(lm_classification))

        lm_regression = LinearModel(Ridge())
        assert (is_regressor(lm_regression))

        parameters = {'kernel': ['linear'], 'C': [1, 10]}
        lm_gs_classification = LinearModel(
            GridSearchCV(svm.SVC(), parameters, cv=2, refit=True, n_jobs=1))
        assert (is_classifier(lm_gs_classification))

        lm_gs_regression = LinearModel(
            GridSearchCV(svm.SVR(), parameters, cv=2, refit=True, n_jobs=1))
        assert (is_regressor(lm_gs_regression))

        # Define a classifier, an invertible transformer and an non-invertible one.

        class Clf(BaseEstimator):
            def fit(self, X, y):
                return self

        class NoInv(TransformerMixin):
            def fit(self, X, y):
                return self

            def transform(self, X):
                return X

        class Inv(NoInv):
            def inverse_transform(self, X):
                return X

        X, y, A = _make_data(n_samples=1000, n_features=3, n_targets=1)

        # I. Test inverse function

[... 285 lines cut ...]
args_msg   = ['copy=True', 'with_mean=True']
extra_args = 2
f          = <function StandardScaler.__init__ at 0x7ffb49156310>
kwargs     = {}
kwonly_args = ['copy', 'with_mean', 'with_std']
sig        = <Signature (self, *, copy=True, with_mean=True, with_std=True)>

/usr/lib64/python3.9/site-packages/sklearn/utils/validation.py:68: FutureWarning
----------------------------- Captured stdout call -----------------------------
Opening raw data file /builddir/build/BUILD/mne-python-0.20.4/mne/decoding/tests/../../io/tests/data/test_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 40199 =     42.956 ...    66.930 secs
Ready.
Current compensation grade : 0
15 matching events found
Applying baseline correction (mode: mean)
Not setting metadata
3 projection items activated
Loading data for 15 events and 421 original time points ...
1 bad epochs dropped

```

cbrnr, 2020-06-03:
This is already fixed in master (not sure about 0.20.5 though).

sanjayankur31, 2020-06-03:
Great. I'll test these with each new release when I update the Fedora package. Thanks.

closed 2020-06-03 by sanjayankur31