mne-tools/mne-python #7861: mne.preprocessing.current_cour

mservantufc, 2020-06-03:
#### Describe the bug
〈Replace this text with a clear and concise description of the bug.〉

#### Steps to reproduce
〈Replace this text with a code snippet or minimal working example [MWE] to
replicate your problem, using one of the [built-in datasets], preferably the
one called [sample]. If you can't replicate on a built-in dataset, provide also
a link to a small, anonymized portion of your data that does yield the error.〉

[MWE]: https://en.wikipedia.org/wiki/Minimal_Working_Example
[built-in datasets]: https://mne.tools/dev/overview/datasets_index.html
[sample]: https://mne.tools/dev/overview/datasets_index.html#sample

#### Expected results
〈Replace this text with a clear and concise description of what you expected to
happen.〉

#### Actual results
〈Replace this text with the actual output, traceback, screenshot, or other
description of the results.〉

#### Additional information
〈Replace this text with the output of `mne.sys_info()`.〉

closed 2020-06-03 by mservantufc

agramfort, 2020-06-03:
@mservantufc you can just edit the issue after opening it without closing.

mservantufc, 2020-06-03:
Sorry wrong manipulation is it possible to delete this post (I reposted a clean version)