mne-tools/mne-python #7867: phantom 4DBTi data includes a bunch of hidden macOS files

drammock, 2020-06-05:
After downloading all the example data while setting up a new system, I noticed that a `__MACOSX` folder ended up inside `mne_data`.  It contains a bunch of finder / spotlight indexing files for the phantom 4DBTi dataset. The `__MACOSX` folder is not listed in the manifest on [the OSF repo](https://osf.io/v2brw/), but is present in the zip file.  In examining the OSF repo, I also noticed `.dropbox` and `.DS_Store` files inside the `MNE-phantom-4DBTi` folder, as well as a zero-byte text file called `Icon` (I dunno if this last one is a mac-generated file or a legitimate part of the dataset?)

agramfort, 2020-06-05:
it's a mac thing. You can clean this up !

>

drammock, 2020-06-05:
OK, I've requested access to MNE's OSF project. Once you grant that, I can fix.

larsoner, 2020-06-05:
Granted

PR #7870 (2020-06-06): MRG, MAINT: update 4DBTi dataset version and hash
closes #7867
closed 2020-06-06 by agramfort