mne-tools/mne-python #7868: Blank Assertion Error when running ICA

barlehmann, 2020-06-05:
[image]

barlehmann, 2020-06-05:
Not sure if the output from running the above came out fully in the above so am pasting it again below: 
[image]

barlehmann, 2020-06-05:
copying last part of output since for some reason that is not showing up:

[image]

agramfort, 2020-06-05:
what version of mne are you using?

mne.sys_info()

gives what?

can you share the file?

>

barlehmann, 2020-06-05:
Thanks for your response. I believe it is version .21: This is the output
from what you suggested:

Platform: Linux-4.19.104+-x86_64-with-Ubuntu-18.04-bionic Python: 3.6.9
(default, Apr 18 2020, 01:56:04) [GCC 8.4.0] Executable: /usr/bin/python3
CPU: x86_64: 2 cores Memory: 12.7 GB mne: 0.21.dev0 numpy: 1.18.4
{blas=openblas, lapack=openblas} scipy: 1.4.1 matplotlib: 3.2.1
{backend=module://ipykernel.pylab.backend_inline} sklearn: 0.22.2.post1
numba: 0.48.0 nibabel: 3.0.2 cupy: Not found pandas: 1.0.4 dipy: Not found
mayavi: Not found pyvista: Not found vtk: Not found

agramfort, 2020-06-05:
you are using a dev version on 0.21

please upgrade to latest stable version or current master version

barlehmann, 2020-06-05:
Thank you for the suggestion - It is now at version mne: 0.20.5, but the
same issue continues to arise.

agramfort, 2020-06-06:
then I need to get the script and the data to replicate

dropbox or google drive or whatever works

>

barlehmann, 2020-06-06:
Thank you Alexandre,

the link to the google colabs ipynb file I am sharing with you is:
https://colab.research.google.com/drive/1e8FB-KdJpFybloglXvWAaaMXsQai9meD?usp=sharing
The piece of code is the 2nd to last piece of code from the bottom with the
text (ICA) right above this section of code. The other sections needed to
be run before this are the frist two bits of code from the top of this code
sheet for installing MNE, and perhaps connecting drive.

The .edf file is being shared at:
https://drive.google.com/open?id=1l75b4f2msaCy6i-hDl-Q9Ogh6ZIa0vDZ

Let me know if you have any further questions, and I'll look forward to
hearing from you,
Bar

agramfort, 2020-06-07:
you are setting n_components = 25 while you only have 19
channels. n_components cannot exceed 19.

feel free to close if it solves the problem

>

barlehmann, 2020-06-07:
That fixed it, thank you very much for your swift and on-point assistance!

larsoner, 2020-06-07:
Reopening as I think we should catch and raise an informative error

agramfort, 2020-06-07:
PR welcome

PR #7879 (2020-06-08): MRG, ENH: Better error messages for ICA
Closes #7868
closed 2020-06-08 by larsoner