mne-tools/mne-python #7871: Forward model from Brainstorm to MNE

vagechirkov, 2020-06-06:
Dear All,

I would like to recreate in MNE exactly the same head model as in Brainstorm. In Brainstorm I am using the following settings to compute the head model: cortex 15000V (tess_cortex_pial_low.mat) source model and overlapping spheres forward modeling method. Is it possible to directly convert a head model from Brainstorm to MNE including co-registration information? If not, what settings should be used to recreate the equivalent head model in MNE?

Looking forward for any advice or suggestion,
Valerii

agramfort, 2020-06-06:
MNE has no implementation of the overlapping sphere model. I would suggest you a single layer BEM model is should work as well or better.

vagechirkov, 2020-06-08:
Dear Alexandre,

thank you for the response. Is it possible to directly import co-registration information from Brainstorm to MNE?

agramfort, 2020-06-08:
I have never done it.

vagechirkov, 2020-06-16:
I have tried the following code to transfer the coregistration information from Brainstorm to MNE: 

```python
from mne.transforms import rotation
from mne.coreg import fit_matched_points
import scipy.io as sio

info = mne.io.read_info(raw_path)
head_lpa = info['dig'][0]['r']
head_nasion = info['dig'][1]['r']
head_rpa = info['dig'][2]['r']
head_pts = np.vstack((head_lpa, head_nasion, head_rpa))

fmri_bst = bst_db + '/anat/CASE/subjectimage_T1.mat'
mri_bst = sio.loadmat(fmri_bst)
origin = mri_bst['InitTransf'][0, 1][:3,3]
mri_bst_lpa = np.array(mri_bst['SCS']['LPA'][0][0][0], dtype=np.float64) + origin
mri_bst_nasion = np.array(mri_bst['SCS']['NAS'][0][0][0], dtype=np.float64) + origin
mri_bst_rpa = np.array(mri_bst['SCS']['RPA'][0][0][0], dtype=np.float64) + origin
mri_pts = np.vstack((mri_bst_lpa, mri_bst_nasion, mri_bst_rpa))

parameters = np.array((0., 0., 0., 0., 0., 0., 1., 1., 1.))
est = fit_matched_points(mri_pts, head_pts, out='params')
parameters[:6] = est
trans = rotation(*parameters[:3])
trans[:3, 3] = np.array(parameters[3:6])
trans = mne.transforms.Transform('mri', 'head', trans)

```

It seems to me that it works correctly. Did I miss something? I have attached alignment plots from Brainstorm and MNE.

[image]
[image]

agramfort, 2020-06-16:
nice. Indeed it seems to work !

>

larsoner, 2020-06-16:
Indeed it seems to work! I'll go ahead and close since it seems like there are no changes needed at the MNE end

closed 2020-06-16 by larsoner