mne-tools/mne-python #7872: EDF+ Annotation Timestamps missing submillisecond accuracy

greydongilmore, 2020-06-06:
It was noticed that the onset time for annotations in an EDF+ file might be missing the submillisecond accuracy accociated with the start time of the recording. This results in annotation timestamps being delayed from the true event time. 

#### Expected results

The EDF+ file used here contains patient data and cannot be shared. However, if this bug is confirmed all EDF+ files will be affected. 

If the EDF+ file is brought into [EDFbrowser](https://www.teuniz.net/edfbrowser/) it will be noticed that the millisecond aspect of the start time is displayed in the `Start` field:

[image]

For instance, this results in the annotation timestamps being:

[image]

#### Actual results

When loading the data into MNE the annotation timestamps return:

[image]

If you take the difference between the MNE timestamp and the EDFbrowser timestamp it is equal to the submillisecond portion of the recording start time.

#### Additional information

I reached out to Teunis van Beelen, the developer of [EDFbrowser](https://www.teuniz.net/edfbrowser/), to get an explanation. The explanation provided was as follows:

> EDF does not have support for subsecond precision of the starttime but EDF+ does. Because EDF+ needed to be backwards compatible with EDF and EDF doesn't have the possibility to write subsecond precision for the starttime into the header, the subsecond part of the starttime is written outside of the EDF header. The way the subsecond part of the starttime is stored in EDF+ is described here: https://www.edfplus.info/specs/edfplus.html#timekeeping

Teunis van Beelen provided a quote from the EDF specs:

> "The startdate/time of a file is specified in the EDF+ header fields 'startdate of recording' and 'starttime of recording'. These fields must indicate the absolute second in which the start of the first data record falls. So, the first TAL in the first data record always starts with +0.X2020, indicating that the first data record starts a fraction, X, of a second after the startdate/time that is specified in the EDF+ header. If X=0, then the .X may be omitted."

In other words, the subsecond part of the starttime is written in the EDF Annotations "signal".

agramfort, 2020-06-07:
do you know any public file that would allow us to
replicate/investigate this?

otherwise you can try to hack the edf.py file. Relevant code starts here:

https://github.com/mne-tools/mne-python/blob/master/mne/io/edf/edf.py#L231

greydongilmore, 2020-06-07:
I have de-identified the file I used:
https://drive.google.com/file/d/1yql0bMGcXXrEZr1At_Cp-eMi6rTKSr_F/view?usp=sharing

agramfort, 2020-06-07:
can you try this PR https://github.com/mne-tools/mne-python/pull/7875

your explanations helped a lot !

>

greydongilmore, 2020-06-08:
Glad I could contribute. The explanations provided by Teunis van Beelen helped me to understand the issue so credit should be given to him.

closed 2020-06-08 by greydongilmore

greydongilmore, 2020-06-08:
@agramfort @cbrnr I tested the response of FieldTrip and I am getting the same behavior. Seems to be missing the submillisecond accuracy:
[image]

Will reference this PR on their repo

cbrnr, 2020-06-08:
Biosig also doesn't support this, I've reported this to the maintainer. Seems like edflib is the only app with proper support (but now also MNE).

greydongilmore, 2020-06-08:
@cbrnr interestingly though [pyEDFlib](https://github.com/holgern/pyedflib/issues/79) is a python wrapper for EDFlib and it also shows the incorrect behavior... just tested

skjerns, 2020-06-09:
@greydongilmore this is because we haven't integrated the new EDFlib version 1.17 yet (see https://github.com/holgern/pyedflib/pull/81)

I hope I find time soon to implement the changes.

PR #7875 (2020-06-08): FIX : EDF+ Annotation Timestamps missing sub-second accuracy
closes #7872