mne-tools/mne-python #7892: Using average referenced forward solutions in MNE

jdue, 2020-06-12:
Dear MNE developers

I would like to do EEG inverse modeling in MNE with some FEM forward solutions which I have pre-computed. The FEM solutions, however, have an *average* reference and from what I can tell your BEM solutions are infinity referenced. Would you expect any problems using an average referenced forward solution with the inverse modeling provided by MNE? The thing is, we would like to make this FEM implementation available for the EEG community by allowing the user to export the solution to MNE.

As an initial test, I did a very simple source simulation and compared inverse solutions (dSPM) based on BEM (computed with MNE) and FEM (pre-computed) with similar geometry and they seem to agree well, however, I would like to hear if would expect this difference in how the fields are referenced to give problems (e.g., depending on the particular inverse method)?

Best,
Jesper

agramfort, 2020-06-12:
they agree because our inverse code does the average referencing internally.

average reference is applied during source imaging and not when computing
the forward

>

jdue, 2020-06-13:
cool, thanks!

closed 2020-06-13 by jdue