mne-tools/mne-python #7901: Rename `master` branch

hoechenberger, 2020-06-15:
Hello,

amid the global [BLM protests](https://en.wikipedia.org/wiki/Black_Lives_Matter), it seems a number of free/open-source software projects are now [reconsidering](https://www.zdnet.com/article/github-to-replace-master-with-alternative-term-to-avoid-slavery-references/) the naming of the main source-code management branch, historically called `master` in `git`. Microsoft seems to be [actively working](https://twitter.com/natfriedman/status/1271253144442253312) to bringing out-of-the-box support for other naming schemes GitHub.

Despite its widespread use in development and hardware interfaces, the term "master" **may** be considered offensive by some. While one could argue that this perception entirely disregards context, I would argue that nothing is lost by dropping the name. I would consider moving to a different name as a token of support for the struggles of ethnic minorities and marginalized people; and it would be a public display of some (granted, cheap, but still visible) efforts to actively embrace an inclusive community. Lastly I would argue that our Code of Conduct actually doesn't leave us much of a choice anyway, as it clearly [states](https://github.com/mne-tools/.github/blob/master/CODE_OF_CONDUCT.md#our-standards) we vow to use "welcoming and inclusive language". Now if the `master` term has been identified to be non-inclusive – regardless of whether **we** feel that this is true or not – it should be dropped.

Looking forward to reading your thoughts on this!

Best wishes,

–Richard

cbrnr, 2020-06-15:
By all means change it if it makes people happy. GitHub will soon default to `main` anyway. However, I think that in this case the term master has nothing to do with master/slave (yes I know the BitKeeper background, but in a plain Git repo there is a master branch but no slave branch(es)). Same for blacklisting/whitelisting, which actually originates from the term black book (which were actual black books).

I'd only caution to wait until we know all implications of that name change. I've read that it can be non-trivial in some cases (e.g. it seems to be hard-coded for the docs), so let's do it only after GitHub releases detailed instructions how to do it.

hoechenberger, 2020-06-15:
Hi @cbrnr 
> However, I think that in this case the term master has nothing to do with master/slave (yes I know the BitKeeper background, but in a plain Git repo there is a master branch but no slave [...]

Yes I do agree, my perception is similar. But others seem to disagree, and I don't mind making a small change to them them feel more welcome ☺️

> I'd only caution to wait until we know all implications of that name change. I've read that it can be non-trivial in some cases (e.g. it seems to be hard-coded for the docs), so let's do it only after GitHub releases detailed instructions how to do it.

Yes absolutely, just wanted to raise this question now to get a small discussion going / get an idea of what others think.

larsoner, 2020-06-15:
Internally I don't think this change would be too difficult, but we should wait until GitHub makes it easy to redirect branch names to do this (last I looked it was not possible to redirect individual branches, only entire repos). Projects that rely on using `https://github.com/mne-tools/mne-python/zipball/master` or similar should have an opportunity to change to the new naming rather than just having things suddenly break.

agramfort, 2020-06-16:
I am tempted to follow here. If core projects that are models for us do
this then we could follow.

GuillaumeFavelier, 2020-07-06:
Not directly related but I should probably rename the `time_viewer` selected as reference in `LinkViewer`:

https://github.com/mne-tools/mne-python/blob/master/mne/viz/_brain/_timeviewer.py#L1221-L1222

hoechenberger, 2020-07-06:
> Not directly related but I should probably rename the `time_viewer` selected as reference in `LinkViewer`:
> 
> https://github.com/mne-tools/mne-python/blob/master/mne/viz/_brain/_timeviewer.py#L1221-L1222

Heh. Do we have a good alternative? `main`, perhaps?

GuillaumeFavelier, 2020-07-06:
`main` or just simply `reference` should do the trick?

hoechenberger, 2020-07-06:
> `main` or just simply `reference` should do the trick?

How about
```python
        controller = self.time_viewers[0]  # select the first time viewer, which controls the scene
        value = controller.time_call.slider_rep.GetValue()
```

cbrnr, 2020-07-06:
Feel free to rename, but master without slaves should be fine (because masters can also have apprentices).

GuillaumeFavelier, 2020-07-06:
Thanks all for the suggestions, I decided to rename into `leader` because it suits the behaviour perfectly. And sorry for the noise on the issue.

hoechenberger, 2020-07-11:
The Linux kernel developers just [decided](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=49decddd39e5f6132ccd7d9fdc3d7c470b0061bb) the use the following terms:

```
+For symbol names and documentation, avoid introducing new usage of
+'master / slave' (or 'slave' independent of 'master') and 'blacklist /
+whitelist'.
+
+Recommended replacements for 'master / slave' are:
+    '{primary,main} / {secondary,replica,subordinate}'
+    '{initiator,requester} / {target,responder}'
+    '{controller,host} / {device,worker,proxy}'
+    'leader / follower'
+    'director / performer'
+
+Recommended replacements for 'blacklist/whitelist' are:
+    'denylist / allowlist'
+    'blocklist / passlist'
+
+Exceptions for introducing new usage is to maintain a userspace ABI/API,
+or when updating code for an existing (as of 2020) hardware or protocol
+specification that mandates those terms. For new specifications
+translate specification usage of the terminology to the kernel coding
+standard where possible.
```

cbrnr, 2020-07-22:
See https://github.com/github/renaming/

hoechenberger, 2020-08-29:
Just for reference, the VS Code Python Extension is chiming in now, too:
https://github.com/microsoft/vscode-python/issues/13645

cbrnr, 2021-01-15:
Should we go for it? https://docs.github.com/en/github/administering-a-repository/renaming-a-branch

hoechenberger, 2021-01-16:
> Should we go for it? https://docs.github.com/en/github/administering-a-repository/renaming-a-branch

I just came here to ask the same, do: yes, let's do that!!

agramfort, 2021-01-16:
have you done with any other project ? with one with many forks like mne?

Let's not experiment with the mne-python repo

>

cbrnr, 2021-01-16:
What about starting with `mne-testing-data`? I've prepared all changes for this repo in #8758.

agramfort, 2021-01-16:
I'll just speak for myself. I have tons of stuff to do and I need to
prioritize things.
Doing things like this is very low priority right now.

So if you want to do it fine. But I will try **not** to look and let you in
charge.

>

cbrnr, 2021-01-16:
Same, this is also low priority for me. I don't want to break anything, so I'd rather not do it unless others look very carefully at the changes.

hoechenberger, 2021-01-16:
The GH docs specifically say that all existing references to the old branch names will continue to work:

```
If you rename a branch, GitHub will automatically redirect links on GitHub.com that contain the old branch name to the equivalent link on the renamed branch. GitHub will also update branch protection policies, as well as the base branch for open pull requests and draft releases.
```

cbrnr, 2021-01-16:
Yes, it should work without #8758. But what if it doesn't? Are there any larger projects that have already switched? I checked numpy, matplotlib, pandas, scipy, scikit-learn - they are all still using master.

hoechenberger, 2021-01-16:
> Yes, it should work without #8758. But what if it doesn't? Are there any larger projects that have already switched? I checked numpy, matplotlib, pandas, scipy, scikit-learn - they are all still using master.

If it doesn't, we just rename it back. I don't understand the hesitation honestly…

hoechenberger, 2021-01-16:
> Are there any larger projects that have already switched? I

vscode-python has switched: https://github.com/microsoft/vscode-python/issues/13645

hoechenberger, 2021-01-16:
I just renamed `master` to `main` on https://github.com/hoechenberger/questplus

After that I had to run locally:
```
git branch -m master main
git fetch origin
git branch -u origin/main main
```
All working now.

cbrnr, 2021-01-16:
@hoechenberger and I will rename the branch this evening.

drammock, 2021-01-16:
I'd like to request that you do this next weekend instead of today, and that you announce it on the mailing list and forum first. Advance warning may avoid a lot of surprises and complaints.

cbrnr, 2021-01-16:
OK, no problem!

cbrnr, 2021-01-17:
I just renamed the master branch to main in MNELAB, everything is fine (also locally).

cbrnr, 2021-01-23:
Done - see https://mne.discourse.group/t/we-have-renamed-the-main-development-branch-on-github-to-main/2526!

PR #8758 (2021-01-25): Replace master -> main
Fixes #7901.
closed 2021-01-24 by cbrnr