mne-tools/mne-python #7903: Array incompatibility error in computing EEG connectivity

barlehmann, 2020-06-15:
[image]

barlehmann, 2020-06-15:
Not sure why the last bit of code was cut off from above:

[image]

drammock, 2020-06-15:
Please do not post *images* of your questions and code.  It is **much easier** if we can copy/paste the code in order to try things out on our systems.

barlehmann, 2020-06-15:
Sorry about that, I was not aware of this, I will certainly be sure not to paste images in the future.

I am pasting my non-image code above again here below: 

import numpy as np
from scipy import linalg

import mne
from mne import io
from mne.connectivity import spectral_connectivity
from mne.datasets import sample

print(doc)
raw = mne.io.read_raw_edf(raw_data_file)

print(raw.info['dig'])
ch_names = ['fp1', 'fp2', 'f3', 'f4', 'c3', 'c4', 'p3', 'p4', 'o1', 'o2', 'f7', 'f8', 't3', 't4', 't5', 't6', 'fz', 'cz', 'pz']
ch_types = ['eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg','eeg']
info = mne.create_info(ch_names=ch_names, sfreq=256, ch_types=ch_types)
raw.info = info
print(raw.info)

montage_kind = "standard_1020"
montage = mne.channels.make_standard_montage(montage_kind)
raw.set_montage(montage, match_case=False, raise_if_subset=True)

print(raw.info['dig'])

events = mne.make_fixed_length_events(raw, start=0, stop=50, duration=2.)
epochs = mne.Epochs(raw, events, event_id=None, tmin=-0.2, tmax=0.5, baseline=(None, 0), picks=picks, preload=False, reject=None, flat=None, proj=True, decim=1, reject_tmin=None, reject_tmax=None, detrend=None, on_missing='ignore', reject_by_annotation=True, metadata=None, event_repeated='error', verbose=None)  

#Compute connectivity for band
fmin, fmax = 3., 9.
sfreq = raw.info['sfreq'] # the sampling frequency
tmin = 5.0 # exclude the baseline period
con, freqs, times, n_epochs, n_tapers = spectral_connectivity(epochs, method='coh', mode='multitaper', sfreq=sfreq, fmin=fmin, fmax=fmax, faverage=True, tmin=tmin, mt_adaptive=False, n_jobs=1)

agramfort, 2020-06-16:
please share a script that runs and replicates the error you want to report

you can use https://gist.github.com/ to share code snippets

barlehmann, 2020-06-16:
@agramfort  - Sorry about that - I might not have included the MNE installation or all the dependencies for the above file: I just pasted the relevant snippet into gist (below), and verified that it brings about the same problem. The only other issue you might have is that the script does not include my .edf file that is being used - let me know how I should get that one to you - I could share it if I have your email address perhaps? 

Also,  upon some further assessment, it appears that the problem occurs before the connectivity code begins, it is in fact a problem at the level of epoching. Any suggestions or thoughts would be greatly appreciated. 

https://gist.github.com/barlehmann/29eb6982115b3000539db5ea5d76c157

agramfort, 2020-06-16:
how do you expect me to debug your script if I don't have access to your
google drive and your file?

>

barlehmann, 2020-06-16:
https://drive.google.com/open?id=1wwS268nX7XVbqLe0f--nA0K80hhd2q5d

Above is the link to my drive's .edf file  -   'AusEC.edf' ,  I am sorry to
burden you with any difficulty, I am still getting familiar with this
process by which others are evaluating my code on git issues (I thought
perhaps someone might see an issue standing out without needing to run the
specific file used). Also, I was hesitant to post a link to my drive on a
public space, so thought I might email it to you, but that is alright, I
just posted it above.

agramfort, 2020-06-16:
`raw.info = info` was a bad idea. See:

```
import mne

raw_data_file = 'AusEC.edf'

raw = mne.io.read_raw_edf(raw_data_file)
raw.rename_channels(lambda s: s.split(' ')[1].split('-')[0])

montage_kind = "standard_1020"
montage = mne.channels.make_standard_montage(montage_kind)
raw.set_montage(montage, match_case=False)

print(raw.info['dig'])

# Pick EEG
picks = mne.pick_types(raw.info, meg=False, eeg=True)

events = mne.make_fixed_length_events(raw, start=0, stop=50, duration=2.)
epochs = mne.Epochs(raw, events, event_id=None, tmin=-0.2, tmax=0.5,
                    baseline=(None, 0), picks=picks)

raw.plot()
```

for a clean solution

closed 2020-06-16 by agramfort

barlehmann, 2020-06-17:
@agramfort Thank you very much for your help - that makes a big difference for the problem with the two arrays, and another problem I was having in a totally different area of the code as well.  Your support has been a meaningful step forward, however, I am still not getting the coherence map to show up after I include a bit of the code from the senor space connectivity tutorial- and am getting the following error messages: 

If I am running the latest (unstable) version:

---> 68 plot_sensors_connectivity(epochs.info, con[:, :, 0])
AttributeError: 'NoneType' object has no attribute '_Renderer'

If I am running the latest (stable) version:

---> 68 plot_sensors_connectivity(epochs.info, con[:, :, 0])
RuntimeError: Could not load any valid 3D backend: ['mayavi', 'pyvista']

A couple of my thoughts on where the error is coming from, that I am going to be looking into more:
1) I am using google colabs to run this code and perhaps the graphing capacities are limited there.
2) I am still becoming familiar with the process of coherence calculation, and perhaps am overlooking an inconsistent argument in the connectivity function. 
3) I am not properly understanding how to load mayavi or pyvista and need to research this.

The gist code snippet of the my error is:
https://gist.github.com/barlehmann/45289573ccbbace46b0d6fa56fe0c5dd

The link to my .edf file on the drive that the code refers to is:
https://drive.google.com/open?id=1wwS268nX7XVbqLe0f--nA0K80hhd2q5d

I will certainly appreciate any tips or suggestions.

barlehmann, 2020-06-17:
@agramfort I can post this last posting in another issue if that is preferable.  I saw that you had closed the issue, and though this last posting of mine had relevance to this issue so it might be useful to post here, but certainly can delete this last posting and move it to a different issue if that is preferred.

drammock, 2020-06-17:
> @agramfort I can post this last posting in another issue if that is preferable.

Even better would be to use the MNE mailing list or the gitter channel.  The GitHub issue tracker is supposed to be reserved for requests for new features and/or bugs in MNE-Python itself (not bugs in your own code).  See https://mne.tools/dev/overview/get_help.html for links.

barlehmann, 2020-06-17:
@drammock - Certainly, will do, sorry about this misunderstanding, I am new to using MNE and had not seen that page before.  This page you mention makes that very clear -sorry again to cause this confusion and the extra work for you guys.  I will certainly post my questions in that way from now on.