mne-tools/mne-python #7911: BUG: inconsistent auto-clim handling for SourceEstimate and VectorSourceEstimate

drammock, 2020-06-19:
In theory, these two things should yield identical results:

```
vecstc = apply_inverse(..., pick_ori='vector')
vecstc.plot(..., brain_alpha=1., overlay_alpha=1., vector_alpha=0.)
```
versus
```
stc = apply_inverse(..., pick_ori=None)
stc.plot(..., surface='white')
```

In practice, `plot_vector_source_estimates` calls `brain.scale_data_colormap` under the hood, while `plot_source_estimates` does not, so the automatic choosing of `clim` is quite different in the two cases.  If you pass in a `clim` dict with your own values, then the two approaches indeed look identical, so probably (?) all that is needed is to remove the call to `brain.scale_data_colormap`.

agramfort, 2020-06-20:
agreed

GuillaumeFavelier, 2020-06-29:
I'm afraid `scale_data_colormap` might be need in `plot_vector_source_estimates` to update the colormap of the glyphs.

larsoner, 2020-06-29:
Where? For PySurfer or PyVista? With my latest PR, I'd be surprised if it's the case for PyVista. If it is, then it seems like a bug. (And if things aren't working for PySurfer, it seems like probably a bug there, too, that should be fixed.)

GuillaumeFavelier, 2020-06-29:
PySurfer, I think it's related to an issue opened a long time ago...

GuillaumeFavelier, 2020-06-29:
I think I found it: https://github.com/mne-tools/mne-python/pull/6510

larsoner, 2020-06-29:
So it sounds like the bug was not completely fixed

PR #8202 (2020-09-03): MRG, BUG: Fix STC limit bug
Close #7911

On `master`, auto control points for the vec STC are `[1.86911522 2.26944519 6.37046324]` and standard STC are `[ 3.95048065  4.56941314 17.72451438]`. On this PR, there are two fixes:

1. `vector_alpha` actually has an effect
2. `_process_clim` uses the `.magnitude()` data for vector mode, which makes more sense.

On this PR, this code:

<details>

```
import os.path as op
import mne

data_path = mne.datasets.sample.data_path()
sample_dir = op.join(data_path, 'MEG', 'sample')
subjects_dir = op.join(data_path, 'subjects')
inv = mne.minimum_norm.read_inv [... cut]
closed 2020-09-03 by GuillaumeFavelier