mne-tools/mne-python #7913: [fNIRS] Misleading error message when the raw_data path does not exist

Swy7ch, 2020-06-22:
#### Describe the bug

When the path provided by the user does not exist (i.e. when the directory does not exist), the error message raised by the **RAWNirx** `__init__` function is 
```
RuntimeError: Expect one evt file, got 0
```

I think it should be something like 
```
RuntimeError: The path you specified does not exist
```
since the error is not about the existence of a file, but rather about the directory's

#### Steps to reproduce
Create a **RAWNirx** instance by reading the data of a non-existent directory :
```
fnirs_raw_dir = 'a/path/that/doesnt/exist'

raw_intensity = mne.io.read_raw_nirx(fnirs_raw_dir,
                                         verbose=True).load_data()
```

#### Expected results
The error message tells the user the path to the specified directory does not exist.

#### Actual results
The error message tells the user the specified directory lacks an `*.evt` file (since it is the first file extension that is checked) but not that the directory does not exist: the existence of the directory is not checked, which is misleading the user.

#### Additional information
Platform:        Windows-10-10.0.18362-SP0
Python:          3.8.3 (default, May 19 2020, 06:50:17) [MSC v.1916 64 bit (AMD64)]
Executable:     C:\Users\juliend\Anaconda3\envs\mne_dev\python.exe
CPU:              Intel64 Family 6 Model 58 Stepping 9, GenuineIntel: 8 cores
Memory:        15.6 GB

mne:              0.21.dev0
numpy:          1.18.1 {blas=mkl_rt, lapack=mkl_rt}
scipy:             1.4.1
matplotlib:     3.1.3 {backend=Qt5Agg}

sklearn:         0.22.1
numba:         0.49.1
nibabel:         3.1.0
cupy:             Not found
pandas:         1.0.3
dipy:             1.1.1
mayavi:        4.7.2.dev0
pyvista:        0.25.3
vtk:              9.0.0
PyQt5:         5.9.2

\-\-

Please note that adding
```
if not op.exists(fname):
    raise RuntimeError('The path you specified does not exist.')
```
or
```
if not op.isdir(fname):
    raise RuntimeError('The path you specified does not exist.')
```
at the start of `__init` function of **RAWNirx** does not lead to raising the error, if *fname* is a non-existent path; there must be a "good" reason for that. I'm in to correct the behaviour if you think it should be fixed.

larsoner, 2020-06-22:
This does seem like something that should be fixed, but let's wait to hear from @rob-luke to see if he agrees or has something else in mind first

rob-luke, 2020-06-23:
Sounds good. I added this code snippet to https://github.com/mne-tools/mne-python/pull/7905

Thanks @Swy7ch for taking the time to report these issues and the neat fixes

Swy7ch, 2020-06-23:
Does the snippet work on your end @rob-luke ? I should have made it clearer that it **doesn't** work for me, but I don't know why : even if I give a random path to the function, the error raised is still about the *.evt file

rob-luke, 2020-06-23:
And @Swy7ch please feel free to review #7905 to make sure it meets your needs

PR #7905 (2020-06-23): WIP: Warn if untested NIRX device
#### What does this implement/fix?
@cbrnr confirmed in #7894 that the NIRX format for saving data is not consistent across devices. I have only added test data for the NIRScout device I have access too. The function now throws a warning if used with an untested device.

#### Additional information
Next step is to improve the reader function to work with these alternative formats. But this warning seems prudent until someone with access to these devices adds reader support.

Closes #7913
closed 2020-06-23 by larsoner