mne-tools/mne-python #7915: Cluster-based Permutation test for Time-Frequency Data - 3D clusters

SebastianSpeer, 2020-06-22:
Hello MNE-team,

Eric Larson has asked me to post this as an issue here.

I've been trying to run cluster-based permutation tests on time-frequency data considering all channels and different frequency ranges across time. 
Colleagues of mine have informed me that Fieldtrip in MATLAB provides an algorithm that allows to search for adjacent frequencies, timepoints and channels that show similar modulations, which then result in 3D clusters. In the examples on statistical analysis of time-frequency data provided on the MNE website one channel only is selected. Is there an option to run a similar analysis that runs over all channels simultaneously and produces 3D clusters? If so is there an example that can be found somewhere and are there suggestions for plotting these outputs? Thank you very much for your help!

Best,

Sebastian

larsoner, 2020-06-22:
Can you try #7916 ?

SebastianSpeer, 2020-06-22:
What should I try? Sorry, not quite sure what to do...

drammock, 2020-06-22:
@SebastianSpeer you would need to (temporarily) use the version of MNE-Python located at https://github.com/larsoner/mne-python/tree/conn

The way to do that is to follow the [using current master](https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master) instructions (but choosing a different name for the environment so you can easily go back to current stable MNE), then installing in that environment the version of MNE-Python from @larsoner's branch where he implemented the new functionality.  You can do this with `pip install git+https://github.com/larsoner/mne-python@conn`.  Then, test away on your data!

Or, if you have a clone of the current master repo installed using `pip install -e`, then you can add @larsoner's fork as an additional remote (`git remote add larsoner https://github.com/larsoner/mne-python`), checkout the `conn` branch (git checkout -b conn larsoner/conn).  Then, test away on your data!

SebastianSpeer, 2020-06-23:
@drammock Thanks that worked fine. 
@larsoner your implementation works fine for me. Thanks a lot. The only question remaining now is how can I detect clusters in this three-dimensional space and plot them in a meaningful way? In the example you provide you plot the clusters for a specfic channel only. However, is there a way to detect and visualize clusters across all three dimensions? Something like a joint plot that shows the average TFR map and then a topo plot indicating where these power modulations are strongest at different points in time?

larsoner, 2020-06-23:
Maybe look into the AverageTFR methods like https://mne.tools/dev/generated/mne.time_frequency.AverageTFR.html#mne.time_frequency.AverageTFR.plot to see if any of them would work? IIRC one of them does a topo plot with TFRs at each channel location

closed 2020-06-25 by larsoner

SebastianSpeer, 2020-06-26:
@larsoner that worked well thanks.

I've now tried to run cluster-based permutation tests with TFCE but it crashes because I am running out of memory. When looking at the memory demands they increase very rapidly as soon as I execute the funciton. I've reduced the step size to 1 for testing and it now tries 28 different thresholds. Could this be due to the extra dimension added? 

I've also increased the buffer_size with now change in memory requirement. I've also set a cache directory, which also did not same to change anything.

'''
out_dir='/data/sebastian/EEG/neural_analysis/response_locked_TFR/'
ensure_dir(opj(out_dir,'cache'))
mne.set_cache_dir(opj(out_dir,'cache'))

power_all_cheatable=mne.time_frequency.read_tfrs(opj(out_dir,'Analysis','Cheatable_NonCheatable','pwr_cheatable-tfr.h5'))[0]
power_all_noncheatable=mne.time_frequency.read_tfrs(opj(out_dir,'Analysis','Cheatable_NonCheatable','pwr_noncheatable-tfr.h5'))[0]

tfce           = dict(start=1, step=1)

T_obs, clusters, cluster_p_values, H0 = \
    permutation_cluster_test([power_all_cheatable.data, power_all_noncheatable.data], n_jobs=15,
                             n_permutations=100, threshold=tfce, tail=0, buffer_size=100)

print("--- %s seconds ---" % (time.time() - start_time))
'''

GiovanniGr, 2021-04-29:
Hello MNE-team,

 I tried to install the version described but there are no files in that directory. In fact https://github.com/larsoner/mne-python/tree/conn returns a 404 error. Is it possible to get it again?

Best,
Giovanni

larsoner, 2021-04-29:
that branch was merged in `https://github.com/mne-tools/mne-python/pull/7916`, if you use latest `main` you should be good to go

GiovanniGr, 2021-04-30:
I tried to do a time-frequency permutation_cluster_test over 126 channels for different observations and for two groups.
So I have 4D data (observations, times, frequencies, channels) for two groups of users but it didn't work. 

`adjacency = sparse.csr_matrix(np.zeros((126,126)))
sample = np.zeros((4,100,6,126))

print(sample.shape)
print(adjacency.shape)

F_obs, clusters, cluster_pv, H0 = mne.stats.permutation_cluster_test(X=[sample, sample], adjacency=adjacency)`

the error raised is 

> ValueError: adjacency (len 126) must be of the correct size, i.e. be equal to or evenly divide the number of tests (75600). If adjacency was computed for a source space, try using the fwd["src"] or inv["src"] as some original source space vertices can be excluded during forward computation

at line 606 of "mne/stats/cluster_level.py" file.
(and 75600 is evenly dividable by 126, but "got_times" is different from "n_times" for the dimension more)

It seems like at most there can be three dimensions and not four.
I thought that with this version it was possible to do it.

Is it not?

Best,
Giovanni

PR #7916 (2020-06-25): MRG, ENH: Add arbitrary connectivity for stats
Arguably part of #4859, originally brought up in #5745, resurfaced on the mailing list today.

Closes #7915

@mmagnuski you might have asked about this sort of thing before, interested in reviewing and/or trying?

In theory at some point we could probably exploit a lattice structure to avoid a lot of unnecessary checks (just like we do in the `*_temporal` variants of our stats functions), but this is at least a good start that should allow people to do these sorts of analyses even if they are slower than they could be.