mne-tools/mne-python #7920: generate the colorbar for plot_topomap in subplots

datalw, 2020-06-23:
Goal: I want to generate 8 topomaps in one figure and for each subplot a colorbar - becaue the data has different scale in each subplot. 
Result: in the figure attached.

My code:
`
evoked=[]
fig, ax = plt.subplots(1,8
[image]
)
for frexi in range(len(frequencyBand)): #len(frequencyBand)=8
    evoked.append ( mne.EvokedArray(values[frexi], raw.info) )
    evoked[frexi].plot_topomap(times=0, vmin = 0, time_unit='s',time_format=None, axes = ax[frexi],
                                         cmap='Spectral_r', colorbar = True,show=False)
    ax[frexi].set_title(str(freBandNames[frexi]), fontsize=8)
`

drammock, 2020-06-23:
Your example is not reproducible, so it's a hard to test (you don't include code that defines the variables `values`, `frequencyBand`, `freBandNames`, or `raw`). Next time please provide a [minimal, complete, and verifiable example](https://stackoverflow.com/help/mcve) that demonstrates the problem you are experiencing. If at all possible, your example should use [MNE-Python's "sample" dataset](https://mne.tools/dev/overview/datasets_index.html#sample) or one of the other built-in datasets.

Can you also tell us what version of MNE-Python you are using?  That plotting function was refactored about six weeks ago, so it may work better if you use current master version.  On master, I would expect your code to raise an error, since you're passing only one `Axes` each time you call the `plot_topomap` method, but it will expect 2 axes each time (one for the head, and one for the colorbar).

datalw, 2020-06-23:
> 
> 
> Your example is not reproducible, so it's a hard to test (you don't include code that defines the variables `values`, `frequencyBand`, `freBandNames`, or `raw`). Next time please provide a [minimal, complete, and verifiable example](https://stackoverflow.com/help/mcve) that demonstrates the problem you are experiencing. If at all possible, your example should use [MNE-Python's "sample" dataset](https://mne.tools/dev/overview/datasets_index.html#sample) or one of the other built-in datasets.
> 
> Can you also tell us what version of MNE-Python you are using? That plotting function was refactored about six weeks ago, so it may work better if you use current master version. On master, I would expect your code to raise an error, since you're passing only one `Axes` each time you call the `plot_topomap` method, but it will expect 2 axes each time (one for the head, and one for the colorbar).
> 
> I would start by trying current master, allocating twice as many axes objects (16 instead of 8), and passing in 2 axes each time through your loop.

Thanks a lot for your advice and sorry for the incomplete inquiry. 
I had the version of '0.20.5'. After updating to '0.20.7' and 16 axes instead of 8, I still could not get the result. 
My complete code is as following. To make the codes short, I used the same values (only a single time point hence 1*20 array) twice in order to produce 2 subplots (instead of 8).

```
%matplotlib qt
import mne
import numpy as np
import matplotlib.pyplot as plt

values = np.array([[6.11871843e-06],
       [4.99458919e-06],
       [4.27978549e-06],
       [4.96262736e-06],
       [3.56721672e-06],
       [3.09460620e-06],
       [3.27792100e-06],
       [2.71426545e-06],
       [4.61115950e-06],
       [3.70161128e-06],
       [4.05230608e-06],
       [3.79710019e-06],
       [1.99561725e-06],
       [2.86023095e-06],
       [2.19611443e-06],
       [3.29457325e-06],
       [4.76445047e-06],
       [3.04804786e-06],
       [2.89658643e-06],
       [3.78034672e-06]])

ch_names_1020sys = ['Fp1', 'Fp2','F3','F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'A2']
info = mne.create_info(ch_names=ch_names_1020sys, sfreq=256, ch_types='eeg')

evokedArr =  mne.EvokedArray(values, info) 
ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')
evokedArr.set_montage(ten_twenty_montage)

fig, ax = plt.subplots(1,4)
evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s',time_format=None, axes = ax[0:1], cmap='Spectral_r', colorbar = True,show=False) #here I also tried axes=ax[0:2], which actually passes 2 axes, but I get error 'Axes and times must be equal in sizes.'
ax[0].set_title('Delta')
evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s',time_format=None, axes = ax[2:3], cmap='Spectral_r', colorbar = True,show=False) #here as well I tried axes = ax[2:4], also error like above
ax[2].set_title('Theta')
```
Result:
[image]

drammock, 2020-06-23:
Thanks for providing reproducible code.  If I run your code in an ipython terminal (and add `plt.show()` at the end) this is what I get:

[image]

As I mentioned before, the changes to this function were only made about 6 weeks ago, so you have to use **current master** (0.21dev0) rather than **current stable** (0.20.7) to get the results I did. See the [instructions for using the development version](https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master).

Note that if you don't like the size of the colorbars being so wide in my version, you can use `gridspec_kw`, see the (slightly edited) version of your code I used:

```python
import mne
import numpy as np
import matplotlib.pyplot as plt

values = np.array([6.11871843e-06, 4.99458919e-06, 4.27978549e-06,
                   4.96262736e-06, 3.56721672e-06, 3.09460620e-06,
                   3.27792100e-06, 2.71426545e-06, 4.61115950e-06,
                   3.70161128e-06, 4.05230608e-06, 3.79710019e-06,
                   1.99561725e-06, 2.86023095e-06, 2.19611443e-06,
                   3.29457325e-06, 4.76445047e-06, 3.04804786e-06,
                   2.89658643e-06, 3.78034672e-06])[:, np.newaxis]

ch_names_1020sys = ['Fp1', 'Fp2','F3','F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2',
                    'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'A2']

info = mne.create_info(ch_names=ch_names_1020sys, sfreq=256, ch_types='eeg')

evokedArr =  mne.EvokedArray(values, info) 
ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')
evokedArr.set_montage(ten_twenty_montage)

fig, ax = plt.subplots(1, 4, gridspec_kw=dict(width_ratios=[5, 1, 5, 1]))

evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s', time_format=None,
                       axes = ax[:2], cmap='Spectral_r', colorbar=True,
                       show=False)
ax[0].set_title('Delta')

evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s', time_format=None,
                       axes = ax[2:], cmap='Spectral_r', colorbar = True,
                       show=False)
ax[2].set_title('Theta')
plt.show()
```

datalw, 2020-06-24:
> 
> 
> Thanks for providing reproducible code. If I run your code in an ipython terminal (and add `plt.show()` at the end) this is what I get:
> 
> [image]
> 
> As I mentioned before, the changes to this function were only made about 6 weeks ago, so you have to use **current master** (0.21dev0) rather than **current stable** (0.20.7) to get the results I did. See the [instructions for using the development version](https://mne.tools/dev/install/advanced.html#using-the-development-version-of-mne-python-latest-master).
> 
> Note that if you don't like the size of the colorbars being so wide in my version, you can use `gridspec_kw`, see the (slightly edited) version of your code I used:
> 
> ```python
> import mne
> import numpy as np
> import matplotlib.pyplot as plt
> 
> values = np.array([6.11871843e-06, 4.99458919e-06, 4.27978549e-06,
>                    4.96262736e-06, 3.56721672e-06, 3.09460620e-06,
>                    3.27792100e-06, 2.71426545e-06, 4.61115950e-06,
>                    3.70161128e-06, 4.05230608e-06, 3.79710019e-06,
>                    1.99561725e-06, 2.86023095e-06, 2.19611443e-06,
>                    3.29457325e-06, 4.76445047e-06, 3.04804786e-06,
>                    2.89658643e-06, 3.78034672e-06])[:, np.newaxis]
> 
> ch_names_1020sys = ['Fp1', 'Fp2','F3','F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2',
>                     'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'A2']
> 
> info = mne.create_info(ch_names=ch_names_1020sys, sfreq=256, ch_types='eeg')
> 
> evokedArr =  mne.EvokedArray(values, info) 
> ten_twenty_montage = mne.channels.make_standard_montage('standard_1020')
> evokedArr.set_montage(ten_twenty_montage)
> 
> fig, ax = plt.subplots(1, 4, gridspec_kw=dict(width_ratios=[5, 1, 5, 1]))
> 
> evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s', time_format=None,
>                        axes = ax[:2], cmap='Spectral_r', colorbar=True,
>                        show=False)
> ax[0].set_title('Delta')
> 
> evokedArr.data *= 10  # to make clear that the colorbars are different
> evokedArr.plot_topomap(times=0, vmin = 0, time_unit='s', time_format=None,
>                        axes = ax[2:], cmap='Spectral_r', colorbar = True,
>                        show=False)
> ax[2].set_title('Theta')
> plt.show()
> ```
> 
> and the result with `gridspec_kw`:
> 
> [image]

Thanks a lot for the help, I updated to the current master and could get the result that I wanted. 
A small flaw in the plot is that the colorbars in the last row are always incomplete (also as shown in your plot above), the same situation after I defined the figure size in the first place. Therefore I have to adjust the plot using fig.subplots_adjust(bottom=0.1). It would be great if the colorbar are complete by default ; )

drammock, 2020-06-24:
>  It would be great if the colorbar are complete by default ; )

I agree.  It's part of our roadmap (#7751) for the coming year to fix a lot of these inconsistencies in the 2D plotting functions.

closed 2020-06-24 by drammock