mne-tools/mne-python #7938: animate_topomap() without drawing contours?

PappaBu, 2020-06-28:
Is it possible to create an animation using [`animate_topomap()`](https://mne.tools/stable/generated/mne.Evoked.html#mne.Evoked.animate_topomap) without drawing contours?
I can hide contours in [`plot_topomap()`](https://mne.tools/stable/generated/mne.Evoked.html#mne.Evoked.plot_topomap) with option `contours=0`, but I can not find something like this in the `animate_topomap()`.
Is there a way around this limitation?

agramfort, 2020-06-28:
yes this option is not implemented. Pull request very welcome to add this.

>

PappaBu, 2020-07-01:
Sorry, but I'm new and just an ordinary user. How can I ask to add this option to the MNE-Python?

drammock, 2020-07-02:
@PappaBu this issue will serve as a feature request.  No guarantees on how soon it will get added though.

sruthisureshkumar-arch, 2026-10-22:
ran this on the current  version 1.14.0.dev121+gf3e0db815 and tested how the contour lines changed on setting the contours to 0 and then 6. here's what i used and the effects on changing the value assigned. 
```python
import numpy as np
import matplotlib.pyplot as plt
import mne

montage = mne.channels.make_standard_montage("biosemi32")
info = mne.create_info(montage.ch_names, 1000.0, "eeg")
info.set_montage(montage)
data = np.random.default_rng(0).standard_normal((32, 100)) * 1e-6
evoked = mne.EvokedArray(data, info)

fig, anim = evoked.animate_topomap(times=evoked.times[::20], contours=0)
plt.show()
```
this is with contours=6 with the lines clearly shown

[image]
this is with contours=0 with the lines no longer visible

[image]

as you can see with contours=0 there are no lines, and with contours=6 the lines show up
p.s: it seems like this issue has been targeted already but not closed - in case it hasn't heres a small contribution from a beginner!

drammock, 2026-10-22:
closed by #13905

closed 2026-10-22 by drammock