mne-tools/mne-python #7941: Error while trying to read EDF file - not enough values to unpack (expected 3, got 0)

data-RanDan, 2020-06-29:
Hi
This is my first time working with EDFs using Python.
This is the error i get when trying to open them.
not enough values to unpack (expected 3, got 0)

I couldn't find much information on-line, I saw that you answered a similar question in the past.
any idea how can i solve the problem?

agramfort, 2020-06-29:
can you share a file to replicate the problem?

>

data-RanDan, 2020-06-29:
[https://drive.google.com/drive/folders/1xJMhtwTYT73VGvZQQwbzMGXozAjgWI40?usp=sharing](url)

cbrnr, 2020-06-29:
The date seems to be the problem:

```pytb
ValueError                                Traceback (most recent call last)
<ipython-input-4-aedc95ac2e76> in <module>
----> 1 raw = mne.io.read_raw_edf("/Users/clemens/Downloads/aaa_00001.edf", preload=True)

~/Repositories/mne-python/mne/io/edf/edf.py in read_raw_edf(input_fname, eog, misc, stim_channel, exclude, preload, verbose)
   1179         raise NotImplementedError(
   1180             'Only EDF files are supported by read_raw_edf, got %s' % (ext,))
-> 1181     return RawEDF(input_fname=input_fname, eog=eog, misc=misc,
   1182                   stim_channel=stim_channel, exclude=exclude, preload=preload,
   1183                   verbose=verbose)

<decorator-gen-163> in __init__(self, input_fname, eog, misc, stim_channel, exclude, preload, verbose)

~/Repositories/mne-python/mne/io/edf/edf.py in __init__(self, input_fname, eog, misc, stim_channel, exclude, preload, verbose)
    111         logger.info('Extracting EDF parameters from {}...'.format(input_fname))
    112         input_fname = os.path.abspath(input_fname)
--> 113         info, edf_info, orig_units = _get_info(input_fname,
    114                                                stim_channel, eog, misc,
    115                                                exclude, preload)

~/Repositories/mne-python/mne/io/edf/edf.py in _get_info(fname, stim_channel, eog, misc, exclude, preload)
    375     misc = misc if misc is not None else []
    376 
--> 377     edf_info, orig_units = _read_header(fname, exclude)
    378 
    379     # XXX: `tal_ch_names` to pass to `_check_stim_channel` should be computed

~/Repositories/mne-python/mne/io/edf/edf.py in _read_header(fname, exclude)
    362     logger.info('%s file detected' % ext.upper())
    363     if ext in ('bdf', 'edf'):
--> 364         return _read_edf_header(fname, exclude)
    365     elif ext in ('gdf'):
    366         return _read_gdf_header(fname, exclude), None

~/Repositories/mne-python/mne/io/edf/edf.py in _read_edf_header(fname, exclude)
    546         meas_id['recording_id'] = fid.read(80).decode('latin-1').strip(' \x00')
    547 
--> 548         day, month, year = [int(x) for x in
    549                             re.findall(r'(\d+)', fid.read(8).decode())]
    550         hour, minute, sec = [int(x) for x in

ValueError: not enough values to unpack (expected 3, got 0)
```

The file doesn't seem to be a valid EDF file though, EDFbrowser can't open it either and gives the following error:

[image]

cbrnr, 2020-06-29:
And opening the file in a hex editor confirms that this is definitely not a valid EDF file:

```
{
HeaderID = EH:000001:000000:000000 ;
ByteOrder = LowByteFirst ;
DataType = SignedInteger ;
Size = 4092732 ;
Dim_1 = 981 ;
Dim_2 = 1043 ;
Image = 0 ;
acq_frame_nb = 0 ;
time = Mon Mar 12 11:25:07 2018 ;
time_of_day = 1520850307.453593 ;
time_of_frame = -12345678901234567168.000000 ;
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                 }
```

Could this actually be some other file format that was (accidentally) renamed to .edf?

woolgathering, 2020-10-28:
This error occurs when exporting post-processed EDF and EDF+ files from WinEEG. I figure the solution is to write some sort of header validator that will put in arbitrary but acceptable values.

cbrnr, 2020-10-28:
The file I saw is not EDF (as in https://www.edfplus.info/), so adding some header fields won't help. Do your files look similar to the one shared in this thread?

hoechenberger, 2021-02-14:
Closing due to inactivity and seemingly not a bug at our end.

closed 2021-02-14 by hoechenberger