mne-tools/mne-python #7942: Function plot incorrectly exit

YeZiyi1998, 2020-06-29:
Here is my code.
###
import os
import mne
import ssl

# This restores the same behavior as before.
# context = ssl._create_unverified_context()
# response = urllib.request.urlopen("https://no-valid-cert", context=context)

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_filt-0-40_raw.fif')
event_fname = os.path.join(sample_data_folder,'MEG','sample','sample_audvis_filt-0-40_raw-eve.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
# print(raw.info)
# raw.plot_psd(fmax=50)
raw.plot(duration=1, n_channels=3)
###
The result is 
###
Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)
###
and there is not expected graph output.

agramfort, 2020-06-29:
can you report what `mne.sys_info()` outputs?

YeZiyi1998, 2020-06-29:
Yes, Here it is.
```
Platform:      Darwin-19.5.0-x86_64-i386-64bit
Python:        3.7.2 (default, Jan 13 2019, 12:50:01)  [Clang 10.0.0 (clang-1000.11.45.5)]
Executable:    /usr/local/opt/python/bin/python3.7
CPU:           i386: 8 cores
Memory:        Unavailable (requires "psutil" package)
mne:           0.20.7
numpy:         1.16.1 {blas=openblas, lapack=openblas}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=MacOSX}

sklearn:       0.20.3
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
None

```

YeZiyi1998, 2020-06-29:
I tried to install psutil and after that the result is:
```
Platform:      Darwin-19.5.0-x86_64-i386-64bit
Python:        3.7.2 (default, Jan 13 2019, 12:50:01)  [Clang 10.0.0 (clang-1000.11.45.5)]
Executable:    /usr/local/opt/python/bin/python3.7
CPU:           i386: 8 cores
Memory:        16.0 GB

mne:           0.20.7
numpy:         1.16.1 {blas=openblas, lapack=openblas}
scipy:         1.2.1
matplotlib:    3.0.3 {backend=MacOSX}

sklearn:       0.20.3
numba:         Not found
nibabel:       Not found
cupy:          Not found
pandas:        1.0.3
dipy:          Not found
mayavi:        Not found
pyvista:       Not found
vtk:           Not found
None
```
However, still have the problem of "Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)".

larsoner, 2020-06-29:
Probably another problem with the macOS backend. Can you try setting the env var `MPLBACKEND=Qt5Agg` or doing `import matplotlib; matplotlib.use('Qt5Agg')` and see if it gets rid of the segfault?

YeZiyi1998, 2020-06-29:
The segfault problem is solved but there is not image of 'raw.plot(duration=1, n_channels=3)' showed, while 'raw.plot_psd(fmax=50)' can show image accurately. Why is this problem occur?

cbrnr, 2020-06-29:
Works for me with the default `osx` backend. @YeZiyi1998 try adding `block=True` in your call to `raw.plot`.

YeZiyi1998, 2020-06-29:
Thanks. Adding "block=True" really works. @cbrnr

closed 2020-06-29 by cbrnr